PX4/PX4-Autopilot #2561: Landed Mode Transition Causes Throttle Spikes

jgoppert, 2015-07-14:
# Zoomed Out

* You can see on a rising or falling edge of the landed transition a throttle spike of about 10% is caused.

[image]

# Zoomed In

* The throttle appears saturated when not in landed mode.
* When in landed mode, the throttle is only briefly at the increased value and then is again saturated.

[image]

To get the OUT0_Out0 on the same scale I just used this normalization (out0 - 1000)/1000

Related commits:
commit 3873fe1c0005950efb197a6fa8cc523bcf87a590
Merge: 2909793 3c3abc0
Author: Lorenz Meier <lm@inf.ethz.ch>
Date:   Sun Jul 12 17:23:20 2015 +0200

    Merge branch 'beta'

...skipping...
    MC land detector: Slightly decrease allowed vertical motion during landed state. This is important so that fast descends do not result in a

commit 5549d480fd55817262cb70ebff299a641285c2c0
Author: Lorenz Meier <lm@inf.ethz.ch>
Date:   Mon Jun 29 10:16:17 2015 +0200

    MC land detector: Update params and fix docs. Allow more motion during the landed state, but become more picky on throttle.

commit 9a3658836154bea32f0f02b3695fd3deddbcbadb
Author: Lorenz Meier <lm@inf.ethz.ch>
Date:   Mon Jun 29 10:15:27 2015 +0200

    MC land detector: If no position information is available, rely on the armed state exclusively to infer the landed condition.

commit 5982eaaf34f53accbf8040c81e5c0ece3ed6487b

LorenzMeier, 2015-07-14:
I'm not sure what this issue report is implying. What is the problem you're trying to solve?

jgoppert, 2015-07-14:
The throttle jumping to 10% immediately after arming and going back down, then spiking again 2 seconds later makes the motors sound irratic. It isn't safety critical but sudden motor spikes shouldn't happen .

LorenzMeier, 2015-07-14:
I haven't seen this with the default position estimators. Have you cross checked this is not a regression from the estimator?

jgoppert, 2015-07-14:
I have not checked with inav or ekf position estimaters yet, but can confirm the mentioned patches introduced this new  bug when land detector is interacting with the lpe estimator in live testing. Has anyone else flight tested master recently to report on the existence of this bug with other position estimators? I will try to track it down tomorrow.

LorenzMeier, 2015-07-14:
What is your MPC_THR_MIN value and your LNDMC_THR_MAX value? The first value should be smaller than the 2nd.

jgoppert, 2015-07-14:
MPC_THR_MIN: 0.1
LNDMC_THR_MAX: 0.2

We test flew more today. Everything worked well except this bug. Getting ready to look at logs now and see if I can figure anything else out.

LorenzMeier, 2015-07-14:
I did more code review but no smoking gun yet. It doesn't look like a threshold thing but more like a state transition reset issue.

LorenzMeier, 2015-07-14:
So I rather suspect the root cause to sit somewhere in the mc_pos_ctrl app.

AndreasAntener, 2015-07-15:
We experience something similar, we can't yet put our finger on it (beta from yesterday). Right after arming we sometimes get a takeoff detection:

[image]

[image]

LorenzMeier, 2015-07-15:
@jgoppert I've started a code review which resulted in this style cleanup, but I don't think that the PR fixes the issue. I did fix one corner case where the detection threshold could be set wrongly.
https://github.com/PX4/Firmware/pull/2566

LorenzMeier, 2015-07-15:
Did any of you come up with new ideas on the root cause? For Ada's log it seems to be the initial swinging of the estimator on boot.

LorenzMeier, 2015-07-16:
James, did you get an opportunity to look further into this? Any new insights?

jgoppert, 2015-07-16:
With better gps signal/trusting lidar more our vertical velocity stayed within the 0.6 m/s bound and we had less transitions. The only odd thing is that arming always causes landed mode to turn off briefly.

SimonWilks, 2015-07-16:
Could the throttling up (spiking) on arming trigger this if it goes beyond what it considers the max throttle for landed detection (LNDMC_THR_MAX)? I just noticed mine is possibly too low. Here is a video of me trying this today without any props on (arming then throttling up slightly): https://goo.gl/photos/XGbZrhsLLtdHA26q8 so throttle alone is enough to push it into this state.

LorenzMeier, 2015-07-17:
Its very hard to debug this without the actual logs. It would be good if you could share the log muncher links along with them.

jgoppert, 2015-07-17:
Oops, I don't store my logs long term in the binary format, here is the csv: https://drive.google.com/file/d/0B3wIEs1gcVahMHI3OHdyaG9ZWEE/view?usp=sharing

LorenzMeier, 2015-07-17:
@AndreasAntener @sjwilks Try increasing LNDMC_ROT_MAX. @jgoppert It seems like your issue is different, but I don't see yet how its triggered.

LorenzMeier, 2015-07-17:
CSV is not very useful to me, the binary tools are much more effective in graphing relevant data fast.

jgoppert, 2015-07-17:
OK, will save them in the future. I like ipython and I find the binary extraction to csv for pandas can be brittle at times.

LorenzMeier, 2015-07-17:
@jgoppert Its the yaw rate which triggered the detector in your log file.

jgoppert, 2015-07-17:
OK, we had some magnetic anomalies that were disturbing the attitude estimate, so that makes sense. We discovered this on our second day of testing.

Ignoring the reason for the landing state being triggered. I think the throttle response to this transition is still not ideal.

jgoppert, 2015-07-17:
If the throttle saturation is just responding to the landing transition, why when it think it has taken off does the throttle reduce down to idle again? (see second plot, 0.5 -2 seconds). If we address this, I will be happy with the current state of the code.

LorenzMeier, 2015-07-22:
I'm closing this, since the testing is looking good. @jgoppert I'm still not convinced its the transition. Please re-test and report back and if it persists please make sure to provide a raw log and preferably another one with the default estimator. I fixed the case reported by Andreas and I have not enough info to go on the case you presented.

closed 2015-07-22 by LorenzMeier