PX4/PX4-Autopilot #2563: Bootloader: Use consistent byte ordering when reading the unique ID

pavel-kirienko, 2015-07-14:
The bootloader and the application use the hardware ID differently: the bootloader uses it as is, but the application swaps the byte order word-wise.

```
066cff54'53498948'87074622'00000000
54ff6c06'48894953'22460787'00000000
```

@LorenzMeier why does the application have to swap the byte order?

LorenzMeier, 2015-07-14:
Not sure, the ST reference manual should have the official details.

davids5, 2015-07-14:
One of the submission I did to nuttx added STM32_SYSMEM_UID, before this, it was #defines in many places that tended to be not portable. As you can see below the address varies from family to family

[image]

but is is always 96 bits.
```
Unique device ID register (96 bits)
The unique device identifier is ideally suited:
• for use as serial numbers (for example USB string serial numbers or other end
applications)
• for use as security keys in order to increase the security of code in Flash memory while
using and combining this unique ID with software cryptographic primitives and
protocols before programming the internal Flash memory
• to activate secure boot processes, etc.
The 96-bit unique device identifier provides a reference number which is unique for any
device and in any context. These bits can never be altered by the user.
The 96-bit unique device identifier can also be **read in single bytes**/half-words/words in
different ways and then be concatenated using a custom algorithm.
```
Using the following code to return the data has the lowest memory footprint.
```
size_t length = 12;
memcpy(hw_version->unique_id, (void *) STM32_SYSMEM_UID, length)
```

@LorenzMeier @tridge Are there any legacy dependencies on the "word-wise" format would it be ok 
to replace the PX4 version with the above?

LorenzMeier, 2015-07-17:
What does ST recommend as the default format? @tridge What implications would a change of the byte order of the unique ID have in terms of log handling?

davids5, 2015-07-17:
From what I have read and code I have seen (macros to return 8,16,32 bits) it is open for interpretation. 

The 96-bit unique device identifier can also be **read in single bytes**/half-words/words in different ways and then be **concatenated using a custom  algorithm**.

davids5, 2015-08-14:
@tridge - any feedback or should we assume it can be changed in px4/firmware to be native ordered - 

memcpy(hw_version->unique_id, (void *) STM32_SYSMEM_UID, length)

tridge, 2015-08-20:
if we changed the format uses in MAVLink telemetry or df logs then that would make it harder to track a users board back to the factory logs. Right now we log numbers like this in the factory:
PX4: 20eb125d NuttX: 7c5ef883
PX4v2 002F0032 35324717 36343032
the 2nd is the stm32 ID. 
We get that string from here: https://github.com/diydrones/ardupilot/blob/master/libraries/AP_HAL_PX4/Util.cpp#L104

tridge, 2015-08-20:
get_board_serial() is here: https://github.com/diydrones/PX4Firmware/blob/master/src/modules/systemlib/board_serial.c#L47

davids5, 2015-11-11:
@LorenzMeier , @pavel-kirienko @tridge FYI: @ksschwabe 

We now have an initiative to present the device's serial number on the usb port from bootloader and the nuttx application. See https://github.com/PX4/Bootloader/pull/41

My first choice would be to reduce the complexity and code needed to that of what we use in the UAVCAN bootloaders:

```
memcpy(hw_version->unique_id, (void *) STM32_SYSMEM_UID, length)
```

Since there is prior use that @tridge discusses above what would you think about adding a function:

```get_SoC_serial_number(uint8_t *dest, uint8_t size)```

That will return the System on Chip Serial number in native format.

Then the UUID can be formed by still calling int get_board_serial(uint8_t *serialid) and the serial number can be retrieved with get_SoC_serial_number.

Both would then appear in the ver command 
UUID  and SerialNumber

Thoughts?

LorenzMeier, 2016-02-21:
Won't fix / change.

closed 2016-02-21 by LorenzMeier