PX4/PX4-Autopilot #2564: No valid mission available error after reboot w/o sending new flight plan

jgoppert, 2015-07-15:
It appears that the mission is not checked and the valid status isn't set until the user manually re-sends the flight plan after reboot.

jgoppert, 2015-07-15:
I've tracked it down to this if statement where the checker is run. Is it possible that mission type is not none when a valid flight plan is still in memory but has not been validated yet?

```c++
/* check if the home position became valid in the meantime */
		if ((_mission_type == MISSION_TYPE_NONE || _mission_type == MISSION_TYPE_OFFBOARD) &&
			!_home_inited && _navigator->home_position_valid()) {
```

LorenzMeier, 2015-07-15:
Not sure - it should be MISSION_TYPE_OFFBOARD since that is where the mission came from.

LorenzMeier, 2015-07-16:
Any new info on this one? I believe its real and I have other reports supporting the findings here.

jgoppert, 2015-07-16:
No new insights. But we confirmed it with another 10 flights or so.

LorenzMeier, 2015-07-17:
Fixed in 253b8f5. Let me know if I missed any corner cases.

commit 253b8f50 (2015-07-17): Load missions correctly after restart. Fixes #2564
closed 2015-07-17 by LorenzMeier