PX4/PX4-Autopilot #2565: New log format

DrTon, 2015-07-15:
Here is description of new log format. It's already implemented and works, but maybe needs some cleanup.

New log format
===============

The log format is self-describing, i.e. parser should know only algorithm of decoding, all information about structure of messages contained in the log.

Features:

 - Message body size up to 256 bytes
 - Message name + fields types/names string size up to 256 bytes
 - Fixed-size arrays support
 - Multi topics support: up to 128 separate streams of one type with "active" marker

Encoding
--------

Log stream consists of messages. Each message represents update on some topic and has following structure:

```
	uint8_t magic_num;		// 0x77
	uint8_t msg_idx;		// Message index
	uint8_t multi_id;		// ID for multi-topic massages
	BODY_TYPE body;			// Message body
```

`magic_num` is always 0x77.

`msg_idx` is message index that defines business message type.

There is special message type that used to describe business messages. It has `msg_idx` = 0 and the following body.

```
	uint8_t idx;			// Message index
	uint8_t size;			// Message size
	uint8_t format_len;		// Size of format string
	uint8_t is_multi;		// Flag if message is multi-topic
	char format_str[256];	// Message description string
```

All structs including body are packed and use **big-endian** byte order.

Message description has following format:

```
<message_name>:<field type> <field_name>;...;<field type> <field_name>;
```

`field_type` is C type. Supported types are:

 - int8_t
 - uint8_t
 - int16_t
 - uint16_t
 - int32_t
 - uint32_t
 - int64_t
 - uint64_t
 - float
 - double
 - bool
 - char

Also supported fixed-size arrays of values: `<type>[<size>]`. Array of `char`s should be interpreted by parser as null-terminated, UTF-8 encoded string.
Some examples:

```
uint8_t mode;
float[3] position;float[3] velocity;
char[128] message;
```

Special messages
----------------

There are some special business messages, that shouldn't be displayed to user but processed by parser in special way.

##### _TIMESTAMP

```
	uint64_t timestamp;
```

`timestamp` is in microseconds. Must be placed before each group of messages.

kd0aij, 2015-07-15:
What is the intended use of the timestamp message? I assume this is to save space in the log by reducing redundancy for groups of messages with identical timestamps.
But the current sdlog2 implementation is throwing away timestamps on critical data such as IMU sensor readings by not writing timestamp messages frequently enough. Does this new format provide a way to guarantee that the timestamps are correct for each and every data sample?

pavel-kirienko, 2015-07-15:
@DrTon @LorenzMeier Does this affect the tasks that @againagainst is about to start working on?

LorenzMeier, 2015-07-15:
Yes. Its probably going to be required in order to make the code generation efficient because it addresses severe limitations of the current logging format.

DrTon, 2015-07-16:
>What is the intended use of the timestamp message? I assume this is to save space in the log by reducing redundancy for groups of messages with identical timestamps.

Exactly, timestamps logic is the same as in sdlog2.

>But the current sdlog2 implementation is throwing away timestamps on critical data such as IMU sensor readings by not writing timestamp messages frequently enough. Does this new format provide a way to guarantee that the timestamps are correct for each and every data sample?

Current sdlog2 implementation writes samples with fixed rate, e.g. 100Hz. If topic rate is e.g. 500Hz then 4 of 5 samples will be skipped anyway and writing "correct" timestamp for each fifth sample will not help, the data will be still not full. I.e. we have snapshots of all topics, and timestamp is time of snapshot, not of the update on individual topic. In new logger implementation we are going to leave this approach (fixed log rate and skipping messages on too frequent topics).

Logging at full rate, without skipping any updates is not possible with current hardware/OS, bitrate is too high and writing eats too much CPU. Also it's not possible with current unbuffered uORB implementation, any small lag in logger and some data will be lost. So it's better to NOT guarantee writing 100% of updates but just write at fixed guaranteed rate.

DrTon, 2015-07-16:
And some thoughts about logger format. This will not change logic but just binary representation of the data. There are following types of the messages:

 - message format description
 - timestamp
 - business message, normal (topic data)
 - business message, multi (for multi-topics, includes instance ID)

Parser must know how to parse each type of the messages, including timestamp. So packing timestamp as business message is useless. The same for message format. And format for normal and multi messages are also a bit different.

What about adding "message type" field to header, e.g. instead of magic num? So we will have these 4 base message types + many types for normal/multi business message. I suspect that generator and parser code will be more clear and short in this case.

kd0aij, 2015-07-16:
I like to do Fourier analysis of things like accelerometer data. This normally relies on a uniform sampling interval, but correct results can be achieved (with some difficulty) using non-uniform sampling. But this is only possible if the timestamps are preserved. That requires including the timestamp in the message.

I see that sdlog2 is using a circular buffer in front of the SD card, similar to my own implementation in MatrixPilot. Experience with MatrixPilot has shown that a 0.25 second buffer is the minimum required to prevent dropouts;  a recent px4log of mine represents about 35K bytes/second throughput, and the default buffer size of 8K is slightly below that minimum requirement.  But I see that 12K bytes is being allocated for fmuv2 at sdlog2 startup in rc.logging, so that looks OK.

For my purposes, it would be desirable to support configuring logging so that the available throughput can be utilized for high fidelity recording of a subset of messages e.g. each and every IMU sensor reading (with the correct individual timestamps).

DrTon, 2015-07-16:
> I like to do Fourier analysis of things like accelerometer data. This normally relies on a uniform sampling interval, but correct results can be achieved (with some difficulty) using non-uniform [...]

>For my purposes, it would be desirable to support configuring logging so that the available throughput can be utilized for high fidelity recording of a subset of messages e.g. each and every IMU sensor reading (with the correct individual timestamps).

Yes, correct individual timestamps will help only if some topic logged at full rate. Anyway if timestamps will be included in topic body you will have both snapshot timeout and individual timeout. But I don't think that it's necessary. Once you will have full rate data for accel topic you can simply assume that sampling is uniform with good precision, ignoring logger timestamps (with some jitter in this case), because actually it is implemented this way in driver. On other hand, if data logged at not full rate you can assume that logger samples at lower but uniform rate :)

DrTon, 2015-07-16:
Logger format reworked, IMO it's more clean now. Still under question:

- Format string: "uint32_t value" looks a bit long, may be use one-char type aliases as in old format ("i value")?
- We need some recommended metadata messages formats for version, parameters etc to make parsers more simple and universal in future.
- Need to think about parameters writing. Currently all parameters written as float, but it would be good to preserve type.

@LorenzMeier, what do you think?

kd0aij, 2015-07-16:
Good point about being able to assume uniform sampling with some jitter.
But what is the use of logging unfiltered sensor data at less than the sampling rate?

DrTon, 2015-07-16:
>But what is the use of logging unfiltered sensor data at less than the sampling rate?
It's still good for simulation, checking of estimators etc. Yes, results will be more noisy, but anyway.

kd0aij, 2015-07-16:
except the "noise" will be aliasing

DrTon, 2015-07-17:
Format updated again. Now log generator (C++11) and parser (Java) implemented, look good, but need some cleanup, will be published soon.

LorenzMeier, 2016-05-22:
On master.

closed 2016-05-22 by LorenzMeier