PX4/PX4-Autopilot #2568: px4cannode-v1: libuavcan pool should be enlarged

pavel-kirienko, 2015-07-15:
@davids5 Pool size in `px4cannode-v1` is insufficient. When the node receives concurrent GetNodeInfo requests, it fails to properly enqueue the responses due to insufficient pool memory available. The issue manifests itself on the bus as outgoing transfers that lack frames in the middle, here's an annotated dump of CAN traffic:

```
(1436992770.657995)  can0  1E3081FD   [8]  23 0D 00 7B 01 00 00 9B   '#..{....' LAST FILE READ REQUEST
(1436992770.657997)  can0  1E3081FD   [8]  2F 66 73 2F 6D 69 63 3B   '/fs/mic;'
(1436992770.657999)  can0  1E3081FD   [8]  72 6F 73 64 2F 66 77 1B   'rosd/fw.'
(1436992770.658000)  can0  1E3081FD   [8]  2F 63 2F 62 33 34 32 3B   '/c/b342;'
(1436992770.658001)  can0  1E3081FD   [8]  31 63 31 34 2E 62 69 1B   '1c14.bi.'
(1436992770.658002)  can0  1E3081FD   [8]  6E 2E 76 61 6C 69 64 7B   'n.valid{'

(1436992770.668997)  can0  1E307D81   [8]  A4 03 00 00 00 00 00 9B   '........' LAST FILE READ RESPONSE
(1436992770.669994)  can0  1E307D81   [8]  00 00 00 00 00 03 1E 3B   '.......;'
(1436992770.669997)  can0  1E307D81   [8]  00 00 00 00 00 00 00 1B   '........'
(1436992770.669998)  can0  1E307D81   [8]  00 00 00 3C 00 02 40 3B   '...<..@;'
(1436992770.680991)  can0  1E307D81   [8]  00 00 00 00 00 00 00 1B   '........'
(1436992770.680994)  can0  1E307D81   [8]  00 04 1F 00 00 00 00 3B   '.......;'
(1436992770.680995)  can0  1E307D81   [8]  00 00 00 00 00 00 50 1B   '......P.'
(1436992770.680997)  can0  1E307D81   [8]  00 02 40 00 00 00 00 3B   '..@....;'
(1436992770.692995)  can0  1E307D81   [8]  00 00 00 00 05 20 00 1B   '..... ..'
(1436992770.692998)  can0  1E307D81   [8]  00 00 00 00 00 00 00 3B   '.......;'
(1436992770.692999)  can0  1E307D81   [8]  00 00 64 00 02 40 00 1B   '..d..@..'
(1436992770.693000)  can0  1E307D81   [8]  00 00 00 00 00 00 00 3B   '.......;'
(1436992770.693987)  can0  1E307D81   [8]  06 21 00 00 00 00 00 1B   '.!......'
(1436992770.693989)  can0  1E307D81   [8]  00 00 00 00 00 78 00 3B   '.....x.;'
(1436992770.693991)  can0  1E307D81   [8]  02 40 00 00 00 00 00 1B   '.@......'
(1436992770.704996)  can0  1E307D81   [4]  00 00 00 7B               '...{'

(1436992770.801993)  can0  10015564   [8]  29 01 00 00 00 00 00 C9   ').......'

(1436992771.182994)  can0  1001557D   [8]  00 00 00 00 08 00 00 C0   '........' FIRST NODE STATUS AFTER UPDATE

(1436992771.223994)  can0  1E01FDE4   [1]  C1                        '.'        GNI REQUEST  100 -> 125
(1436992771.236993)  can0  1E0164FD   [8]  DF 3A 00 00 00 00 00 81   '.:......' GNI RESPONSE 125 -> 100
(1436992771.236995)  can0  1E0164FD   [8]  00 00 00 01 01 7A 80 21   '.....z.!' ...
(1436992771.238006)  can0  1E01FD81   [1]  C1                        '.'        GNI REQUEST  1   -> 125
(1436992771.238009)  can0  1E0164FD   [8]  4B E5 E1 F8 D0 43 BC 01   'K....C..' GNI RESPONSE 125 -> 100
(1436992771.238010)  can0  1E0101FD   [8]  DF 3A 00 00 00 00 00 81   '.:......' GNI RESPONSE 125 -> 1
(1436992771.238011)  can0  1E0101FD   [8]  00 00 00 01 01 7A 80 21   '.....z.!' ...
(1436992771.296997)  can0  1E0101FD   [8]  4B E5 E1 F8 D0 43 BC 01   'K....C..' ...
(1436992771.297000)  can0  1E0101FD   [8]  C6 DF 8E 01 00 06 6C 21   '......l!' ...
(1436992771.297001)  can0  1E0101FD   [8]  FF 54 53 49 89 48 87 01   '.TSI.H..' ...
(1436992771.297996)  can0  1E0101FD   [8]  07 46 22 00 00 00 00 21   '.F"....!' ...
(1436992771.297999)  can0  1E0101FD   [8]  00 6F 72 67 2E 70 69 01   '.org.pi.' ...
(1436992771.298000)  can0  1E0101FD   [8]  78 68 61 77 6B 2E 70 21   'xhawk.p!' ...
(1436992771.298001)  can0  1E0101FD   [8]  78 34 63 61 6E 6E 6F 01   'x4canno.' ...
(1436992771.298002)  can0  1E0101FD   [6]  64 65 2D 76 31 61         'de-v1a'   ...
(1436992771.298004)  can0  1E0164FD   [8]  78 68 61 77 6B 2E 70 21   'xhawk.p!' GNI RESPONSE 125 -> 100
(missing 4 frames)
(1436992771.298997)  can0  1E0164FD   [8]  78 34 63 61 6E 6E 6F 01   'x4canno.' ...
(1436992771.299000)  can0  1E0164FD   [6]  64 65 2D 76 31 61         'de-v1a'   ...

(1436992771.356964)  can0  10040A7D   [8]  B2 2E 00 00 00 00 9A 80   '........' APPLICATION STARTS HERE
(1436992771.356967)  can0  10040A7D   [8]  42 19 14 00 4E 00 00 20   'B...N.. '
(1436992771.357964)  can0  10040A7D   [3]  19 00 40                  '..@'
(1436992771.476975)  can0  10040A7D   [8]  B2 2E 00 00 00 00 9A 81   '........'
(1436992771.476978)  can0  10040A7D   [8]  42 19 14 00 4E 00 00 21   'B...N..!'
(1436992771.476982)  can0  10040A7D   [3]  19 00 41                  '..A'
(1436992771.537010)  can0  10040A7D   [8]  B2 2E 00 00 00 00 9A 82   '........'
(1436992771.537013)  can0  10040A7D   [8]  42 19 14 00 4E 00 00 22   'B...N.."'
(1436992771.537014)  can0  10040A7D   [3]  19 00 42                  '..B'
(1436992771.639000)  can0  10015501   [8]  23 01 00 00 00 00 00 C3   '#.......'
(1436992771.639003)  can0  10040A7D   [8]  B2 2E 00 00 00 00 9A 83   '........'
(1436992771.639004)  can0  10040A7D   [8]  42 19 14 00 4E 00 00 23   'B...N..#'
(1436992771.639005)  can0  10040A7D   [3]  19 00 43                  '..C'
(1436992771.756994)  can0  10040A7D   [8]  B2 2E 00 00 00 00 9A 84   '........'
(1436992771.756997)  can0  10040A7D   [8]  42 19 14 00 4E 00 00 24   'B...N..$'
(1436992771.756999)  can0  10040A7D   [3]  19 00 44                  '..D'
(1436992771.801995)  can0  10015564   [8]  2A 01 00 00 00 00 00 CA   '*.......'
(1436992771.857028)  can0  10040A7D   [8]  B2 2E 00 00 00 00 9A 85   '........'
(1436992771.857031)  can0  10040A7D   [8]  42 19 14 00 4E 00 00 25   'B...N..%'
(1436992771.857032)  can0  10040A7D   [3]  19 00 45                  '..E'
(1436992771.976997)  can0  10040A7D   [8]  B2 2E 00 00 00 00 9A 86   '........'
(1436992771.976999)  can0  10040A7D   [8]  42 19 14 00 4E 00 00 26   'B...N..&'
(1436992771.977001)  can0  10040A7D   [3]  19 00 46                  '..F'
(1436992772.037000)  can0  10040A7D   [8]  B2 2E 00 00 00 00 9A 87   '........'
(1436992772.037003)  can0  10040A7D   [8]  42 19 14 00 4E 00 00 27   'B...N..''
(1436992772.037004)  can0  10040A7D   [3]  19 00 47                  '..G'
(1436992772.157001)  can0  10040A7D   [8]  B2 2E 00 00 00 00 9A 88   '........'
(1436992772.157004)  can0  10040A7D   [8]  42 19 14 00 4E 00 00 28   'B...N..('
(1436992772.157005)  can0  10040A7D   [3]  19 00 48                  '..H'
(1436992772.216999)  can0  1001557D   [8]  01 00 00 00 00 00 00 C1   '........'

(1436992772.224082)  can0  1E01FDE4   [1]  C2                        '.'        GNI REQUEST  100 -> 125
(1436992772.224353)  can0  1E0164FD   [8]  40 76 01 00 00 00 00 82   '@v......' GNI RESPONSE 125 -> 100
(1436992772.224356)  can0  1E0164FD   [8]  00 00 00 01 01 7A 80 22   '.....z."'
(1436992772.276950)  can0  1E0164FD   [8]  4B E5 E1 F8 D0 43 BC 02   'K....C..'
(1436992772.276953)  can0  10040A7D   [8]  B2 2E 00 00 00 00 9A 89   '........'
(1436992772.386960)  can0  1E0164FD   [8]  C6 DF 8E 01 00 06 6C 22   '......l"'
(1436992772.386963)  can0  10040A7D   [8]  B2 2E 00 00 00 00 9A 8A   '........'
(1436992772.386964)  can0  10040A7D   [8]  42 19 14 00 4E 00 00 2A   'B...N..*'
(1436992772.387964)  can0  10040A7D   [3]  19 00 4A                  '..J'
(1436992772.387965)  can0  1E0164FD   [8]  FF 54 53 49 89 48 87 02   '.TSI.H..'
(1436992772.387966)  can0  1E0164FD   [8]  07 46 22 00 00 00 00 22   '.F"...."'
(1436992772.387967)  can0  1E0164FD   [8]  00 6F 72 67 2E 70 69 02   '.org.pi.'
(1436992772.387968)  can0  1E0164FD   [8]  78 68 61 77 6B 2E 70 22   'xhawk.p"'
(1436992772.388945)  can0  1E0164FD   [8]  78 34 63 61 6E 6E 6F 02   'x4canno.'
(1436992772.388947)  can0  1E0164FD   [6]  64 65 2D 76 31 62         'de-v1b'
```

Increasing the pool size to 4K fixes the issue:

```
 Udate:
UAVCAN  getNumFreeBlocks in bytes 3984
              total       used       free    largest
Data:         16000      12600       3400       1808
Prog:        132096     107520      24576      24576
   2 uavcan            532  2044/ 2084 240 (240)  RUN   
```

Also, there's now a better way to monitor pool usage than `getNumFreeBlocks()` - consider [`getPeakNumUsedBlocks()`](https://github.com/UAVCAN/libuavcan/blob/master/libuavcan/include/uavcan/dynamic_memory.hpp#L75).

pavel-kirienko, 2015-07-15:
Testing has been performed on `uavcan_timing_fix` #2567

davids5, 2015-07-15:
@pavel-kirienko 

What is the minimum it can be?

The purpose of this app it to showcase UAVCAN, to help with adoption.
It does provide the uavcan ESC and Indication (albeit hacked for Hz)
There is no real user code other than the LED driver. The goal is to show there will be adequate Flash and RAM resources available to implement the User functionality.  

If all the free ram is used by UAVCAN then adoption will be difficult. 

If the maximum Transfer is ~0.5K What should this be?

pavel-kirienko, 2015-07-16:
> What is the minimum it can be?

Depends on the application. For this particular case the minimum should be approx:

    16 TX frames * 48 bytes per block * 2 = 1536 bytes

Multiplication by 2 is necessary because the library does not allow the TX queue to take more than half of the whole pool.

Since we're talking about a demo app that's better be able to work with the user's logic without any fine-tuning, it makes sense to err towards greater pool size, hence I propose 4K. There's still 3400 bytes left, and note that this is not a bare-bone application - it's running some logic too, which when removed will free some extra memory. I think 4K should be acceptable for a demo.

> If all the free ram is used by UAVCAN then adoption will be difficult.

I want to make a point once again that UAVCAN != libuavcan. Hopefully we'll have libcanard soon, which caters specifically for low-end systems.

davids5, 2015-07-17:
Ok based on our conversation I set it to 2048 I tested with retriver

```
 Udate:
UAVCAN  getPeakNumUsedBlocks() in bytes 480
              total       used       free    largest
Data:         16000      10952       5048       3240
Prog:        132096     107520      24576      24576
   2 uavcan         299991  1984/ 2484 240 (240)  RUN
```

closed 2015-07-17 by davids5