PX4/PX4-Autopilot #2574: Better Portability to New Board Types

ksschwabe, 2015-07-17:
This should probably be marked as an enhancement.

Currently there are a number of modules/files in the Px4 Firmware that use #ifdefs to configure the module for the various boards supported by the Px4 framework. These usually take the form
```
#ifdef CONFIG_ARCH_BOARD_PX4FMU_V1
  <Px4FMU-v1 specific configurations>
#endif
#ifdef CONFIG_ARCH_BOARD_PX4FMU_V2
  <Px4FMU-v2 specific configurations>
#endif 
#ifdef CONFIG_ARCH_BOARD_AEROCORE
  <Aerocore specific configurations>
#endif
```
This hardcodes the board configurations into the source files, so if a new board needs to be configured or supported, the source code needs to be altered. It would be preferable if all board-specific configurations are placed in the src/drivers/boards directories or board_config.h files. This also allows users to port the Px4 Firmware to their custom boards without needing to make alterations to the Px4 Firmware source files.

I have already moved the board-specific ADC configurations to the board_config.h files in pull requests #2552 and #2573, which have subsequently been merged in.

I am now trying to work out how to move the configurations in [drv_gpio.h](https://github.com/PX4/Firmware/blob/master/src/drivers/drv_gpio.h), [fmu.cpp](https://github.com/PX4/Firmware/blob/master/src/drivers/px4fmu/fmu.cpp), [gpio_led.c](https://github.com/PX4/Firmware/blob/master/src/modules/gpio_led/gpio_led.c), and [px4io.cpp](https://github.com/PX4/Firmware/blob/master/src/drivers/px4io/px4io.cpp) over to a more portable framework. Possibly the board-specific functions should be defined in a file in src/drivers/boards directory like what is currently done with defining and setting up the servos.

The full list of files that use these board-specific #ifdefs is:
* [gpio_led.c](https://github.com/PX4/Firmware/blob/master/src/modules/gpio_led/gpio_led.c)
* [commander.cpp](https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp)
* [dsm.c](https://github.com/PX4/Firmware/blob/master/src/modules/px4iofirmware/dsm.c)
* [px4io.h](https://github.com/PX4/Firmware/blob/master/src/modules/px4iofirmware/px4io.h)
* [protocol.h](https://github.com/PX4/Firmware/blob/master/src/modules/px4iofirmware/protocol.h)
* [registers.c](https://github.com/PX4/Firmware/blob/master/src/modules/px4iofirmware/registers.c)
* [sensor_params.c](https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensor_params.c)
* [mtd.c](https://github.com/PX4/Firmware/blob/master/src/systemcmds/mtd/mtd.c)
* [version.h](https://github.com/PX4/Firmware/blob/master/src/lib/version/version.h) - We can probably move the defines here directly over to board_config.h
* [fmu.cpp](https://github.com/PX4/Firmware/blob/master/src/drivers/px4fmu/fmu.cpp)
* [px4io.cpp](https://github.com/PX4/Firmware/blob/master/src/drivers/px4io/px4io.cpp)
* [drv_gpio.h](https://github.com/PX4/Firmware/blob/master/src/drivers/drv_gpio.h)
* [meas_airspeed.cpp](https://github.com/PX4/Firmware/blob/master/src/drivers/meas_airspeed/meas_airspeed.cpp)
* [hmc5883_i2c.cpp](https://github.com/PX4/Firmware/blob/master/src/drivers/hmc5883/hmc5883_i2c.cpp)

Does anyone have any suggestions on how to proceed?

ksschwabe, 2015-07-27:
Related to this, the Pixhawk wiki will have to be updated. Especially https://pixhawk.org/dev/porting, which is already out of date.

LorenzMeier, 2016-12-22:
We are well on track to resolve this completely, so I'm closing it.

closed 2016-12-22 by LorenzMeier