PX4/PX4-Autopilot #2576: Wrong min thrust limit in MANUAL mode for multicopters

DrTon, 2015-07-18:
In MANUAL mode thrust shouldn't be limited, i.e. this line should be removed:
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1419

Now there is large 10% dead zone on thrust control in MANUAL.

Also description of the parameter should be corrected:
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_params.c#L46

>Minimum vertical thrust. It's recommended to set it > 0 to avoid free fall with zero thrust.

It's minimal vertical thrust _in altitude controlled modes_, i.e. all modes > MANUAL, but not MANUAL.

LorenzMeier, 2015-07-18:
I've adjusted the title a bit to reflect the user interfacing discussion this is about, as its not a bug per se.

LorenzMeier, 2015-07-18:
Adressed in https://github.com/PX4/Firmware/pull/2577 (when parameters are set accordingly).

LorenzMeier, 2015-07-22:
Fixed in master.

closed 2015-07-22 by LorenzMeier