PX4/PX4-Autopilot #2578: Q Estimator : Visual heading support

mhkabir, 2015-07-19:
Getting in support for external visual yaw in the new ```attitude_estimator_q``` is essential, since the attitude ekf has been mostly phased out. 

Ideally, we would separately track heading states from the mag and vision, and have expose a parameter to do either of the following :
1 . Use both mag and vision. Vision would correct the estimate differentially, giving us a long term stable and true global heading. This also makes sure that visual tracking going bad isn't going to cause wild yawing in flight. 
2. Disable the mag completely for indoor situations with magnetic interference, like the current approach. 

I should be able to take this up soon-ish depending on testing capabilities.

TSC21, 2015-07-19:
:+1: for this. I was planning to advance to this soon or later, so let me know if you need something ;)

mhkabir, 2015-12-21:
Long-since fixed.

closed 2015-12-21 by mhkabir