PX4/PX4-Autopilot #2584: FW tuning guide clarification

kd0aij, 2015-07-21:
In this section of the FW tuning guide: "Tuning the Trim Offsets with the Integrator Gain"
there is no mention of the TRIM_ROLL,PITCH,YAW parameters. There is also no mention of the SENS_BOARD_X,Y,Z_OFF parameters.

The description of the trim parameters says that by "using the GCS" you may copy RC trims, but it is not clear how this works, or why it is needed.

The description of the board offset parameters says that they are used to correct "misalignment", but it isn't stated what "proper" alignment is, or what the symptoms of misalignment might be.

This guide would be greatly improved by adding a complete description of all (currently unstated) design assumptions and clear definitions of things like 
1) correct orientation of the body frame (autopilot frame?) relative to the airframe
2) cruise attitude and relevant parameters
... other items of which I'm currently ignorant

kd0aij, 2015-07-21:
I'd be happy to do the edits, but I'm not sure I know the correct answers :)
I suppose I could propose changes for review... is there a better mechanism for such proposals than using this issue?

LorenzMeier, 2015-07-22:
> In this section of the FW tuning guide: "Tuning the Trim Offsets with the Integrator Gain"
> there is no mention of the TRIM_ROLL,PITCH,YAW parameters. There is also no mention of the [...]

I wouldn't mention the params since they are easy to get wrong. Instead, these two steps should be mentioned:

  * Put the plane into its level flight position and click on "Level" in the sensor calibration view
  * Fly the plane in manual and trim it, land, hit "copy trim" in the RC calibration view and reset your trims to zero

That will trim to a zero attitude and to zero-rate control surfaces.

> 
> The description of the board offset parameters says that they are used to correct "misalignment", but it isn't stated what "proper" alignment is, or what the symptoms of misalignment might be.
> 

Misalignment of vehicle body frame to autopilot body frame. Symptoms are constant climbing, diving or rolling.

> This guide would be greatly improved by adding a complete description of all (currently unstated) design assumptions and clear definitions of things like 
> 1) correct orientation of the body frame (autopilot frame?) relative to the airframe
> 2) cruise attitude and relevant parameters
> 3) FW_P_ROLLFF parameter
> ... other items of which I'm currently ignorant

Seems like you figured this all out, so it seems you seem perfectly equipped to fix the page up.

kd0aij, 2016-05-25:
Can't believe I didn't put any URLs in here! Not sure if this guide still exists.

closed 2016-08-07 by kd0aij