PX4/PX4-Autopilot #2600: Consider auto-disarming after landing

LorenzMeier, 2015-07-24:
With a suitable landing timeout the system could auto-disarm. This is to assist lazy pilots so they never start a flight armed or let a system sit on the ground armed.

orangelynx, 2015-07-26:
Only if it can be disabled please. Also it would require perfect landing detection.

severinleuenberger, 2015-07-26:
I had a landing 200m away from home point. I did leave the transmitter on  the ground. While walking towards the UAV,  the signal was lost and RTL kicked in. When i arrived my plane,  the Props where already burried  in the dirt and the Esc gone in smoke...  So... YES.  A trustable landed Mode who disarms would be nice. But it should never ever be possible to happend in air during flight...

bansiesta, 2015-07-26:
@severinleuenberger: a working/trusted disarm mode would be enough for your case because there is no RTL when "landed".

What I don't like about auto-disarm is that people just get lazy and stop disarming the vehicles. This means that if the "landed" detection and auto-disarm once doesn't work, it can also be dangerous.

LorenzMeier, 2015-10-21:
@DonLakeFlyer Also wants this.

LorenzMeier, 2015-10-21:
@tumbili Something to look into medium term 8)

DonLakeFlyer, 2015-10-21:
The issue for me is that you have to go back through the switches to get out of Mission mode to then disarm. That took me through AtlCtl down to Manual which can lead to quirkiness if you don't have the thottle on the right place when you do that. Beginner mistake, I know. But stupid-proof would be better.

LorenzMeier, 2015-10-23:
Implemented here. Needs testing.
https://github.com/PX4/Firmware/pull/3056

tumbili, 2015-11-03:
Upper PR is merged, closing.

closed 2015-11-03 by tumbili