PX4/PX4-Autopilot #2611: Altitude control in beta allows full pitch down

AndreasAntener, 2015-07-27:
@LorenzMeier
In altctl/posctl with beta the plane pitches down fully (vertical descent) when I give it full pitch down input. Altitude control kicks in again when I let go, but it has to climb back up again then to what ever altitude it was supposed to hold. Pitching up does not have this effect.

(I flew with #2610)

LorenzMeier, 2015-07-27:
That's expected and won't be fixed for this release. We will re-introduce altitude climb rate limits on the next one.

LorenzMeier, 2015-07-27:
That's expected and will only be fixed on the next release.

LorenzMeier, 2015-07-27:
Could I have the log file from the flight though? I'd love to see if the init issues are gone.

AndreasAntener, 2015-07-27:
What were the initial issues? I also flew hil just before you pushed the fix and it did very weird stuff in posctl, almost tumbling, which I blamed on my setup at first..

http://dash.oznet.ch/view/DRKiL3NJebyAG4GUqtzR5D
[image]

kd0aij, 2015-07-27:
@AndreasAntener Looks like you had a long walk to retrieve that plane!

AndreasAntener, 2015-07-27:
;)

kd0aij, 2015-07-29:
I flew Beta (Firmware Version	75085dc) yesterday on two different planes. Behaved reasonably will on the Ninox, which is now tuned fairly well for pitch control, but I also observed a steep full throttle dive at full down pitch command. This behavior is certainly undesirable, if not a bug in the TECS logic:
[image]

The Acromaster, which is not yet properly tuned, also exhibited the full throttle dive, but zero pitch resulted in fairly rapid altitude gain, making altctl unusable.
[image]

LorenzMeier, 2015-07-29:
@kd0aij I need the log file, I can't make sense of it yet.

kd0aij, 2015-07-29:
Acromaster flight report: https://docs.google.com/document/d/1sAEtfk8iRfWeuABWV2IDBuEOunTv3p-vg-9zigMXScg/edit?usp=sharing

Ninox flight report: https://drive.google.com/open?id=1-h1CpWUT3PzXXIJUXEtE2UIwhBlr5SSotWPyibBkvlE

LorenzMeier, 2015-08-17:
Fixed.

closed 2015-08-17 by LorenzMeier