PX4/PX4-Autopilot #2616: Can't connect to MAVLink via telemetry

austinard, 2015-07-28:
I am using the PX4 v1 flight stack, qgroundcontrol v 2.6.0 (stable), and a Windows 7 OS. I can connect perfectly fine via USB connection, but when I connect via telemetry, the connection terminates itself as soon as qgroundcontrol requests the waypoint list from the board and refuses to reconnect unless I restart the board. This cycle repeats infinitely. When I flash the px4 firmware directly from the eclipse file, the connection problem is solved, but qgroundcontrol refuses to allow calibration and wont show all parameters because the firmware is outdated. At this point, qgroundcontrol forces me to update firmware through qgroundcontrol. If I do this, the telemetry link issue persists and I find myself stuck. Everything was working fine until one day I updated firmware through qgroundcontrol. What are possible sources of this problem?

LorenzMeier, 2015-07-28:
With how many wires have you connected your telemetry setup? Just two or also the RTS / CTS line?

austinard, 2015-07-28:
I am using XBee 802.15 radios and have the adapter for the onboard link connected to the fmu UART2 port. I count 5 wires plugged in: RTS - Flow control into XBee, TX - Serial data from XBee, RX - Serial data into XBee, 5V - power to regulator, and CTS - Flow control from XBee.

austinard, 2015-07-28:
Also, the telemetry connection has worked well over the past couple of months as far as connectivity goes and this problem randomly arose last week. However, I was never able to exchange waypoints or parameters with qgroundcontrol via telemetry, only via USB. I wonder if the two are somehow related?

LorenzMeier, 2015-07-28:
Sounds like it. You need RX, TX and flow control to work in order to communicate successfully. @sjwilks this sounds a bit related to your most recent issue.

AndreasAntener, 2015-07-29:
@LorenzMeier I have the same problem with the V1 board since a while. We'll check the wiring again but all the lines are connected.

LorenzMeier, 2015-07-29:
Do you have 5 wires to the radio modem or just 4?

AndreasAntener, 2015-07-29:
5, also tried different radio and cable the last time

austinard, 2015-07-29:
Update: I have tried wiring up my backup px4 stack with a fresh SD card and was able to connect perfectly via telemetry. However, after I specified the aircraft type (10" quadrotor), the telemetry connection issue came back. I tried picking other random airframes and got the same response from all. I went back to my original px4 stack and tried connecting telemetry with both SD cards and both worked fine until airframe selection.

LorenzMeier, 2015-07-29:
Try reducing this number until it works:
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L477

E.g. try 3000, then 2000 and so on. It would imply that hardware flow control is not operational though, which according to your description should be.

LorenzMeier, 2015-07-31:
Fix pushed to Beta.

LorenzMeier, 2015-08-01:
This is also fixed on stable and master. Sorry for the inconvenience.

closed 2015-08-01 by LorenzMeier