PX4/PX4-Autopilot #2619: Log folder named "sessNNN" with 3D gps fix

kd0aij, 2015-07-29:
log folder "sess004" showed a creation date of 2015-07-28 on the SD card, and creation dates of files within the folder also reflected the correct UTC time

AndreasAntener, 2015-07-31:
Same here today, sessionXXX folders with correct creation/modification date

SimonWilks, 2015-08-23:
@LorenzMeier any ideas on where to look on this one? This is almost all we ever see now. There is often a timestamp on the log file itself so the system seems to have a GPS timestamp but it isn't making it through to the logger.

LorenzMeier, 2015-08-23:
Here is the dir creation:
https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L444-L452

LorenzMeier, 2015-08-23:
GPS timestamp gets copied here:
https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L1335-L1339

And time conversion is here:
https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L410

The system defaults to only use GPS time.

PR #2764 (2015-08-26): Fix usage of GPS time for naming directories and files in sdlog2.
- Don't try to convert gps time to seconds a second time round
- If requested (via the SDLOG_GPSTIME param) only use GPS time when a 3D fix is present.

This fixes #2619
closed 2015-08-26 by LorenzMeier