PX4/PX4-Autopilot #2622: Intermittent Yaw Issues with PX4 Master

Aperture-Laboratories, 2015-07-29:
I have been running into a few issues with yaw drift on a 3DR Quad X and a carbon fiber tube hexacopter. I know the tube frames are looked down upon but we have bolted (drilled through the tubes) everything down to ensure that no rotation of the motors occurs. Upon take off it will rotate about 25-30 degrees to the right, slowly. Those numbers aren't solid, as it varies from flight to flight but it happens in offboard and manual flight. The pixhawk doesn't seem to be correcting (not sure how to check this).

Quad X Specs:
Firmware: Lastest (as of yesterday) Master
Motors: Sunny Sky X3108S-19 / KV: 720
ESCs: 30A Magic ESCs
GPS: 3DR GPS/ Compass Combo
Battery: 10000 mAh 25C QuadroCopter
Power Distribution: 3DR X8 PDB

Hexacopter Specs:
Firmware: Latest (as of yesterday) Master
Motors: Stock 3DR Motors from X8
ESCs: Stock 20A ESCs from 3DR
GPS: Zubax GNSS
Battery: 10000 mAh 25C QuadroCopter
Power Distribution: 3DR X8 PDB

Steps I've taken to mitigate the issue (on both copters, unless specified otherwise):
1. Calibration: I have re-calibrated the whole copter about 5 times. This includes a mag cal. I have tried with the pixhawk's built in magnometer and the 3DR GPS/Compass to no avail. Hexacopter used the Zubax GNSS.

2. I have checked the center of balance (but I believe the copter should try to counter this on its own) and everything was fine.

3. EMI: Re-arranging the ESCs and power distribution board didn't have much of an affect on the amount of interference it would encounter. I put a layer of copper tape beneath the pixhawk and raised the pixhawk one inch above that to reduce any interference.

4. I changed the motors from the stock 3DR ones on the Quad X to the Sunny Sky ones mentioned above an

5. Changed props from all carbon fiber to plastic. Still saw issue so I ordered fresh plastic props, still same issue.

Quad X Logs:
Log1: http://dash.oznet.ch/view/mrMnPh5XBF7k6Jx6fpxqiE
Log2: http://dash.oznet.ch/view/MSVN6JGoHZ5QX5c9KbUouT

Youtube Video 1: https://www.youtube.com/watch?v=sqNg1UggU6o&feature=youtu.be -- Corresponds with log1.

Youtube Video 2: https://www.youtube.com/watch?v=MIZj7m12_fA&feature=youtu.be

If anyone needs more information/logs, let me know.

tumbili, 2015-08-19:
@Aperture-Laboratories Hi, sorry for not writing for so long.
Try to increase the parameters "MC_YAW_P" and see if that helps. I would say you could go to max. 6 but first try smaller values. It seems like your yaw tracking is not too good.
Let me know if that helped.

LorenzMeier, 2017-01-04:
Closing as stale. We improved UAVCAN mag support recently, it might have also been related to two mags publishing to the same topic.

closed 2017-01-04 by LorenzMeier