PX4/PX4-Autopilot #2630: Sensors : independence from boot timing

mhkabir, 2015-07-31:
@LorenzMeier, I have some time to take this up now, and some hints on how to best use the uORB priority parameter would be appreciated.
Links to areas in the sensors app where things need to be done would be ever better :)

LorenzMeier, 2015-07-31:
You can query the priority here:
https://github.com/PX4/Firmware/blob/master/src/modules/uORB/uORB.h#L341

The prio values are here:
https://github.com/PX4/Firmware/blob/master/src/modules/uORB/uORB.h#L68-L76

And any ORB_ADVERTISE_MULTI() call will have one of them associated.

That priority needs to be copied into sensors_combined:
https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L1030

And finally the switching logic in the new switcher needs to obey it:
https://github.com/PX4/Firmware/pull/2441

I would propose you hold off integrating that PR and leave that to me, since there are quite a few changes in that one.

mhkabir, 2015-08-13:
@LorenzMeier Done.

LorenzMeier, 2015-09-21:
This is almost done. What is required is to allow the commander to evaluate the sensor state only when transitioning to armed.

LorenzMeier, 2015-11-06:
I'm closing this. Any UAVCAN related issues need to be debugged by a UAVCAN user / maintainer.

closed 2015-11-06 by LorenzMeier