PX4/PX4-Autopilot #2634: QGroundControl doesn't connect on PX4FMUv1 without IO

webconn, 2015-08-02:
I use PX4FMU without PX4IO (use my PCB for ESC routing), and after last update (on stable and beta) it failed to connect to QGroundControl tool. I erased EEPROM (set default params) and it began to work. But if SYS_USE_IO is set to 0, QGroundControl doesn't show parameters (only "Analyze" tab is working).

It worked with older firmwares. Maybe there are some problems in /etc/init.d/rcS?

DonLakeFlyer, 2015-08-02:
That usually means that all the parameters were not sent to QGC which is a firmware problem.

webconn, 2015-08-02:
Even parameter loading bar in QGC didn't move, so...

DonLakeFlyer, 2015-08-02:
Run with the --full-logging command line options for more details on what is going on.

webconn, 2015-08-02:
I've tried both situations (with SYS_USE_IO set either to 0 or 1). When it is 1, log output looks nice, but with SYS_USE_IO set to 0 I've caught these lines:

... (line no. 2849)
PX4ParameterLoaderLog: Short description: "Failsafe channel mapping"
PX4ParameterLoaderLog: Long description: "The RC mapping index indicates which channel is used for failsafe If 0, whichever channel is mapped to throttle is used otherwise the value indicates the specific rc channel to use"
PX4ParameterLoaderLog: Min: "0"
PX4ParameterLoaderLog: Max: "18"
src/uas/UASWaypointManager.cc 1083 "Requesting waypoint list..."
Unknown message from system: 1 message: 106
ALSA lib pcm_dmix.c:1024:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2267:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2267:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2267:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_dmix.c:1024:(snd_pcm_dmix_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started

DonLakeFlyer, 2015-08-03:
The latest version of QGC (daily build) should tell you at startup if it is missing parameters which should have been sent by the firmware. Are you running that, and if so do you not get a pop-up telling you there was a problem with parameters?

LorenzMeier, 2015-08-04:
The part of the log you shared is a red herring. You will need to share the complete log.

LorenzMeier, 2015-08-04:
This is the true failure: 
```
mixer: can't open /dev/pwm_output1
: No such file or directory
[i] Error loading mixer: /etc/mixers/pass.aux.mix 
pwm: can't open /dev/pwm_output1: No such file or directory
```

LorenzMeier, 2015-08-04:
Fixed in da4d8a5c2b9ff220877caa976671a46a887bdd78.

closed 2015-08-04 by LorenzMeier