PX4/PX4-Autopilot #2637: Battery current is not send via MAVLink

AndreasAntener, 2015-08-04:
@LorenzMeier 
Apart from hardware issues discussed [here](https://groups.google.com/forum/#!topic/px4users/bI1-ZbecoN8), there is also software bug. Yesterday we noticed that we don't get a current reading via MAVLink but it is logged correctly. To be more precise, the current reading we got over MAVLink was 0 or 1.

In https://github.com/PX4/Firmware/blob/0e26c2b37b362c19d30f0544ed6d21bc408bcd7d/src/modules/mavlink/mavlink_messages.cpp#L550 and https://github.com/PX4/Firmware/blob/0e26c2b37b362c19d30f0544ed6d21bc408bcd7d/src/modules/mavlink/mavlink_messages.cpp#L575 the battery current for MAVLink get's divided by 10. The target field is uint16. According to specs it's supposed to contain milliamps * 10. This suggests that the source field contains milliamps, but as far as I can tell it actually contains amps. This would match the effect we saw for currents between 0 and 15 amps.

The scaling for the MAVLink message changed during the last commit.

tnedev, 2015-08-04:
We have the same issue. No current reading, no current consumed through MAVLink.

LorenzMeier, 2015-08-04:
In obvious cases like this (where / 10 should be just changed to * 100) please directly file a pull request.

AndreasAntener, 2015-08-04:
Ok, I wasn't sure why you changed it.

PR #2638 (2015-08-04): Fix current scaling for mavlink message
Fixes #2637
closed 2015-08-04 by LorenzMeier