PX4/PX4-Autopilot #2641: Sending Waypoints over different mavlink links will mess up system

oberion, 2015-08-05:
The px4 code allows for multiple mavlink link. Setting some waypoints over one mavlink link works fine, even the navigator has no problem with it. However, requesting the current waypoints over another mavlink link will fail. More crucially, sending new waypoints over the other mavlink link will write on the active mission list.
The main issue is the _dataman_id variable in the MavlinkMissionManager class storing the active mission list. This variable is defined as an instance variable which is only synchronized in the constructor. If a mavlink link is updating the mission list, it will only update its instance variables. the other instances are not aware of the updated mission!
A dirty hack trying to solve the issue is attached to the end. However i am not quite happy with it, thus i only attach a patchfile. Things that need to be changed:
- All mavlink instances should be aware of the current active list.
  * I have changed the _dataman_id variable from instance type to class type together with the _count and _current_seq variables.

- Only one instance at a time should be able to write a new mission list.
  * another class variable _transfer_in_progress checks for active transfers from other instance
  * This is not really threadsafe, should be guarded.

- The mavlink instances should be aware of mission changes.
  * In the patch each instance is checking for _dataman_id changes
  * There might be a more elegant solution using the uorb system 

- the mavlink instances should inform over their link that the mission has been changed.
  * Here i see the biggest issue. At the moment the mavlink mission messages are not really defined in such a way, that the onboard system can update the receiver about mission changes see http://qgroundcontrol.org/mavlink/waypoint_protocol. Thus i will only send a waypoint_count message as soon as the mission is updated.
   * The proper way would be to define a new mavlink message for telling the receiver that the onboard mission has changed and he shall update it. 
   * If the waypoint_count message is used, QGC needs to be adapted. Until now the state machine on QGC is programmed such that it will ignore the message when it has not send a mission request before! But it will already work for the mavros bridge.

I think the patch might be a good basis to start fixing the issue. the patch should work and was already flight tested. Feel free to integrate the patch into the firmware or modify it.

```
From a985b37c80ebf3827ba14dd518c7be959bc0bd42 Mon Sep 17 00:00:00 2001
From: oberion 
Date: Wed, 5 Aug 2015 16:14:17 +0200
Subject: [PATCH] mavlink/mavlink_mission: Dirty fix for setting waypoints over
 multiple mavlink link

---
 .gitmodules                             |  2 +-
 src/modules/mavlink/mavlink_main.cpp    |  2 ++
 src/modules/mavlink/mavlink_mission.cpp | 58 +++++++++++++++++++++++++--------
 src/modules/mavlink/mavlink_mission.h   | 12 +++++--
 4 files changed, 57 insertions(+), 17 deletions(-)

diff --git a/.gitmodules b/.gitmodules
index b05de95..3e6e75c 100644
--- a/.gitmodules
+++ b/.gitmodules
@@ -4,7 +4,7 @@
 [submodule "NuttX"]
 	path = NuttX
 	url = git://github.com/PX4/NuttX.git
-[submodule "libuavcan"]
+[submodule "src/lib/uavcan"]
 	path = src/lib/uavcan
 	url = git://github.com/UAVCAN/libuavcan.git
 [submodule "Tools/genmsg"]
diff --git a/src/modules/mavlink/mavlink_main.cpp b/src/modules/mavlink/mavlink_main.cpp
index 8901dd0..dfc1afe 100644
--- a/src/modules/mavlink/mavlink_main.cpp
+++ b/src/modules/mavlink/mavlink_main.cpp
@@ -1710,6 +1710,8 @@ Mavlink::task_main(int argc, char *argv[])
 		hrt_abstime t = hrt_absolute_time();
 
 		update_rate_mult();
+		
+		_mission_manager->check_active_mission();
 
 		if (param_sub->update(&param_time, nullptr)) {
 			/* parameters updated */
diff --git a/src/modules/mavlink/mavlink_mission.cpp b/src/modules/mavlink/mavlink_mission.cpp
index bc0b269..2086edb 100644
--- a/src/modules/mavlink/mavlink_mission.cpp
+++ b/src/modules/mavlink/mavlink_mission.cpp
@@ -58,6 +58,12 @@
 #endif
 static const int ERROR = -1;
 
+int MavlinkMissionManager::_dataman_id = 0;
[... 147 lines cut ...]
 
-	int			_dataman_id;				///< Dataman storage ID for active mission
-	unsigned		_count;					///< Count of items in active mission
-	int			_current_seq;				///< Current item sequence in active mission
+	static int		_dataman_id;				///< Global Dataman storage ID for active mission
+	int				_my_dataman_id;				///< class Dataman storage ID
+	static bool		_dataman_init;				///< Dataman initialized
+	
+	static unsigned		_count;					///< Count of items in active mission
+	static int			_current_seq;				///< Current item sequence in active mission
 
 	int			_transfer_dataman_id;			///< Dataman storage ID for current transmission
 	unsigned		_transfer_count;			///< Items count in current transmission
@@ -117,6 +122,7 @@ private:
 	unsigned		_transfer_current_seq;			///< Current item ID for current transmission (-1 means not initialized)
 	unsigned		_transfer_partner_sysid;		///< Partner system ID for current transmission
 	unsigned		_transfer_partner_compid;		///< Partner component ID for current transmission
+	static bool _transfer_in_progress;				///< Global variable checking for current transmission
 
 	int			_offboard_mission_sub;
 	int			_mission_result_sub;
-- 
1.9.5.msysgit.1

```

LorenzMeier, 2015-08-06:
I would appreciate a pull request for this with the flight tested version. I think this is perfectly fine as a stopgap measure.

oberion, 2015-08-07:
Ok, i will make a pull request later today. I will then also make a pull request on the qgc.

LorenzMeier, 2015-08-17:
Thanks for the fix, applied!

closed 2015-08-17 by LorenzMeier