PX4/PX4-Autopilot #2660: Implement "find model" buzzer when crashing / landing out of sight

LorenzMeier, 2015-08-08:
The purpose of this feature is to allow a pilot to retrieve the plane / copter in a corn field or similar environment where its really hard to find the plane. There are a couple of assumptions about this:

  * The user is not using a GCS / tablet. If he would, it would be trivial to just walk to the current location. So one condition would be that no telemetry link is connected or no GPS position is known.
  * The system didn't just land normally. This can mean a couple of things:
    * There were very high accelerations ("crash")
    * The landing detector says the system is landed, and the RC link became lost

The easiest first stab would be to trigger the buzzer when:

  * The plane is landed (condition_landed = true)
  * There is a lost data link or a lost RC link (*important*: A link only becomes lost when it was valid at some point. This is important since we don't want to trigger this if someone is flying without telemetry or flying without RC - only loosing something that we once had should trigger it. This is all implemented, so the right flags only need to be evaluated)

The tune should be unique to tell it apart from error indications.

The right location to implement this is the commander, and it would be good if it would be done in a way to allow an easy port when the commander gets rewritten as C++ class.
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp

vooon, 2015-08-08:
Need way to chose condition when trigger alarm.

E.g. when my plane crash, ESC is pulled away from engine and control wire - no motor beeps or servo power, also external WiFi is smashed. But OBC, Pixhawk and RC receiver is still powered and operational.
I think in that case pixhawk won't enable buzzer because OBC link is still there.

LorenzMeier, 2016-12-26:
Done in PR from @anton-matosov

closed 2016-12-26 by LorenzMeier

anton-matosov, 2017-01-01:
@LorenzMeier  Code in my PR triggers buzzer only in case of failsafe which might be not triggered. In addition if model is disarmed buzzer will not trigger even in case of RC loss as failsafe doesn't get triggered.

It might be also a good idea to have buzzer switch configuration option.

LorenzMeier, 2017-01-01:
PR welcome 8)

anton-matosov, 2017-01-01:
sure