PX4/PX4-Autopilot #2665: ublox: PX4FMU loose connection with  Neo-M8N

webconn, 2015-08-10:
While vehicle is powered on and has a GPS fix (according to LED on GPS module), PX4 may sometimes loose all GPS satellites for a short time (couple of milliseconds, as I think), and renew the connection again. Buzzer sounds like it loose GPS signal, in QGC "number of satellites" value falls to 0 and then return to last value quickly. While all things are working, GPS module see ~15 satellites, this issue was tested on large opened areas.

GPS module is custom, but pretty simple, just UART connection between PX4FMU and U-Blox Neo-M8N-0-01.

All things were correct on commit 364d323 (since July 4).

LorenzMeier, 2015-08-11:
This is the diff since that commit:
```
diff --git a/src/drivers/gps/gps.cpp b/src/drivers/gps/gps.cpp
index 30fbdfd..7971936 100644
--- a/src/drivers/gps/gps.cpp
+++ b/src/drivers/gps/gps.cpp
@@ -310,7 +310,7 @@ GPS::task_main()
 			_report_gps_pos.cog_rad = 0.0f;
 			_report_gps_pos.vel_ned_valid = true;				
 
-			//no time and satellite information simulated
+			/* no time and satellite information simulated */
 
 
 			if (!(_pub_blocked)) {
@@ -351,28 +351,46 @@ GPS::task_main()
 
 			unlock();
 
+			/* the Ashtech driver lies about successful configuration and the
+			 * MTK driver is not well tested, so we really only trust the UBX
+			 * driver for an advance publication
+			 */
 			if (_Helper->configure(_baudrate) == 0) {
 				unlock();
 
-				//Publish initial report that we have access to a GPS
-				//Make sure to clear any stale data in case driver is reset
+				/* reset report */
 				memset(&_report_gps_pos, 0, sizeof(_report_gps_pos));
-				_report_gps_pos.timestamp_position = hrt_absolute_time();
-				_report_gps_pos.timestamp_variance = hrt_absolute_time();
-				_report_gps_pos.timestamp_velocity = hrt_absolute_time();
-				_report_gps_pos.timestamp_time = hrt_absolute_time();
 
-				if (!(_pub_blocked)) {
-					if (_report_gps_pos_pub > 0) {
-						orb_publish(ORB_ID(vehicle_gps_position), _report_gps_pos_pub, &_report_gps_pos);
+				if (_mode == GPS_DRIVER_MODE_UBX) {
+					/* Publish initial report that we have access to a GPS,
+					 * but set all critical state fields to indicate we have
+					 * no valid position lock
+					 */
 
-					} else {
-						_report_gps_pos_pub = orb_advertise(ORB_ID(vehicle_gps_position), &_report_gps_pos);
+					_report_gps_pos.timestamp_time = hrt_absolute_time();
+
+					/* reset the timestamps for data, because we have no data yet */
+					_report_gps_pos.timestamp_position = 0;
+					_report_gps_pos.timestamp_variance = 0;
+					_report_gps_pos.timestamp_velocity = 0;
+
+					/* set a massive variance */
+					_report_gps_pos.eph = 10000.0f;
+					_report_gps_pos.epv = 10000.0f;
+					_report_gps_pos.fix_type = 0;
+
+					if (!(_pub_blocked)) {
+						if (_report_gps_pos_pub > 0) {
+							orb_publish(ORB_ID(vehicle_gps_position), _report_gps_pos_pub, &_report_gps_pos);
+
+						} else {
+							_report_gps_pos_pub = orb_advertise(ORB_ID(vehicle_gps_position), &_report_gps_pos);
+						}
 					}
-				}
 
-				// GPS is obviously detected successfully, reset statistics
-				_Helper->reset_update_rates();
[... 57 lines cut ...]
 		}
@@ -677,16 +677,16 @@ UBX::payload_rx_add_mon_ver(const uint8_t b)
 			// Part 1 complete: decode Part 1 buffer and calculate hash for SW&HW version strings
 			_ubx_version = fnv1_32_str(_buf.payload_rx_mon_ver_part1.swVersion, FNV1_32_INIT);
 			_ubx_version = fnv1_32_str(_buf.payload_rx_mon_ver_part1.hwVersion, _ubx_version);
-			UBX_WARN("VER hash 0x%08x", _ubx_version);
-			UBX_WARN("VER hw  \"%10s\"", _buf.payload_rx_mon_ver_part1.hwVersion);
-			UBX_WARN("VER sw  \"%30s\"", _buf.payload_rx_mon_ver_part1.swVersion);
+			UBX_DEBUG("VER hash 0x%08x", _ubx_version);
+			UBX_DEBUG("VER hw  \"%10s\"", _buf.payload_rx_mon_ver_part1.hwVersion);
+			UBX_DEBUG("VER sw  \"%30s\"", _buf.payload_rx_mon_ver_part1.swVersion);
 		}
 		// fill Part 2 buffer
 		unsigned buf_index = (_rx_payload_index - sizeof(ubx_payload_rx_mon_ver_part1_t)) % sizeof(ubx_payload_rx_mon_ver_part2_t);
 		_buf.raw[buf_index] = b;
 		if (buf_index == sizeof(ubx_payload_rx_mon_ver_part2_t) - 1) {
 			// Part 2 complete: decode Part 2 buffer
-			UBX_WARN("VER ext \" %30s\"", _buf.payload_rx_mon_ver_part2.extension);
+			UBX_DEBUG("VER ext \" %30s\"", _buf.payload_rx_mon_ver_part2.extension);
 		}
 	}
 
```

I'm not sure yet what triggers this, and I would appreciate if you could help debugging it.

LorenzMeier, 2015-11-10:
@tumbili If it works with the Pixfalcon then we can close this.

bansiesta, 2015-11-12:
> Selected commutication protocol: UBX+NMEA.
This seems wrong. Only UBX should be chosen.

I have a PixFalcon here with an M8 and can let it record over night.

LorenzMeier, 2015-11-12:
That would be awesome!

bansiesta, 2015-11-13:
The log ends after 10MB and a bit more than 1000 s. The data looks valid until there but I'll have to repeat the test.

bansiesta, 2015-11-14:
I did an indoor bench test and had a 3D fix over 26 hours straight.
This was with d289ac7ec44f0c8e8d3fd143c5ff565bd5292439 using a PixFalcon and the M8N GPS included with it.

The stock uBlox firmware flashed was: 
```
gps: VER hw  "  00080000"
gps: VER sw  "                  2.01 (75350)"
gps: VER ext "                   2.01 (75331)"
gps: VER ext "                  PROTVER 15.00"
gps: VER ext "           FIS 0xEF4015 (79189)"
gps: VER ext "                  MOD NEO-M8N-0"
gps: VER ext "          GPS;SBAS;GLO;BDS;QZSS"
```

[image]

LorenzMeier, 2015-11-14:
Awesome, thanks! I've fixed a logging related buffer overflow this morning, which could have caused your earlier logging issue. Closing.

closed 2015-11-14 by LorenzMeier