PX4/PX4-Autopilot #2666: ESP8266 Wifi module config support via UART

LorenzMeier, 2015-08-11:
The ESP8266 Wifi module is an ultra-low-cost serial to WIFI bridge ($7.00!) which could serve as a Wifi telemetry link for smartphones. It can be configured as access point: http://tech.scargill.net/esp8266-as-an-access-point/

It can be obtained from here:
https://www.sparkfun.com/products/13678

What we need is:

  * A small AT command parser / config library in src/lib which configures this module
    * SSID
    * UDP mode
    * IP of the GCS
    * baud rate
  * A MAVLink message which contains the SSID and password and sets a flag wether to create this network or join an infrastructure one. There should also be a way to request the existing settings from the GCS side.
  * The MAVLink instance listening on USB could relay the configuration to the MAVLink instance listening on the serial port where the ESP8266 is connected, allowing the GCS to set up the module via USB

@DonLakeFlyer @dogmaphobic I believe spending time on this is much more worthwhile compared to messing with the broken mobile serial port support. Do we have Android / iOS instructions somewhere already? I would love to start to get my hands dirty.

dogmaphobic, 2015-08-13:
Where do you see this running? On the drone side or on the QGC side? I'm not sure I understand how this could be used without some extra hardware.

mhkabir, 2015-08-14:
On the drone side of course. The MAV can host an AP and the rest of our GCS devices can connect to it using their in-built wireless radios.
The only thing I'm concerned about is the range. The internal radios on the tablet/laptop would be the limiting factor.

hkld, 2015-08-15:
Just a comment: To my knowledge, the use of WLAN is not allowed on board of airborne systems. I can check the reference, but it should be written somewhere in the part of the documents, you normally don't read. Honestly speaking, I'm not 100'% sure and also, it could be the decision to ignore this "formality", and yes, I know there are plenty of WLAN based systems out there.

sjwilks, 2015-08-15:
I'd appreciate a reference if you can find one and also where it applies.

hkld, 2015-08-16:
A collegue of mine pointed me to the source ... it's not a general issue, but a German legislation linked to the frequency regulations. It is written that the use of the frequencies 5470 - 5725 MHz for WLAN is not permitted for connections between aircrafts and ground stations ... see the following link (in German), footnote #6
  http://www.bundesnetzagentur.de/SharedDocs/Downloads/DE/Sachgebiete/Telekommunikation/Unternehmen_Institutionen/Frequenzen/Allgemeinzuteilungen/2010_07_WLAN_5GHz_pdf.pdf?__blob=publicationFile
Therefore 2.4GHz is probably not an issue. In Austria the frequencies 5470 – 5750 MHz are allowed for outdoor applications, but there is no limitation concerning applications on aircrafts, see (also in German):
  https://www.rtr.at/de/tk/Spektrum5GHz/1997_bmvit-info-052010de.pdf

LorenzMeier, 2015-12-18:
@dogmaphobic Glad we almost fixed this 8). This is to track the ESP8266 SSID config we discussed.

LorenzMeier, 2016-02-18:
Done via wifi - closing.

closed 2016-02-18 by LorenzMeier