PX4/PX4-Autopilot #2680: PX4_PARAM_DEFINE not passing default values to QGC

mswingtra, 2015-08-12:
The issue is that PX4_PARAM_DEFINE doesn't pass default values to QGC, it works only with PARAM_DEFINE. But px4_param should be used since it works on multiplatform, and having these values in qgc is very useful too.
For me it seems that the problem lies within srcparser.py which doesn't get default values since it's not in the line when px4_param is used, and they are present in the line with param_define.  
Any idea how to solve it?

DonLakeFlyer, 2015-08-12:
I don't understand why you say px4_param should be used since it is multi platform? It is only currently used for something less than 10 params. The version that works with QGC is used for the remaining 400+ params. What is the issue?

mswingtra, 2015-08-12:
On our platform, we use nuttx and posix, and we'll soon implement ros as well. So for that reason, all our custom parameters are defined with px4_param and it'd be great to have their default values in qgc as well.

DonLakeFlyer, 2015-08-13:
SITL compiles using posix and it uses the same PARAM_DEFINE without issue as far as I can tell. Still don’t understand why you would need to use PX4_PARAM_DEFINE.

LorenzMeier, 2015-08-14:
SITL is not using ROS. We have three systems: NuttX, POSIX threads and POSIX ROS.

boosfelm, 2015-08-28:
@mswingtra Can you please provide your fix upstream? Thanks!

LorenzMeier, 2015-08-28:
Thanks for the fix.

closed 2015-08-28 by LorenzMeier