PX4/PX4-Autopilot #2688: Logging somethimes not working on master.

tumbili, 2015-08-13:
While testing the VTOL code we realized that many logs (not all, just the important ones) are corrupted.
Flightplot complains about e.g. "Unknown message type: 47". The unknown message type is not always the same.

LorenzMeier, 2015-08-13:
There is no message with index 47.

LorenzMeier, 2015-08-13:
@tumbili Please cross-test this branch.

AndreasAntener, 2015-08-13:
Flightplot has difficulties always after the same messages, for example:
```
Unknown message type: 63
Message before null: VISN 17236521
Message before that: ATTC 17236521

Unknown message type: 61
Message before null: VISN 68439553
Message before that: ATTC 68439553

Unknown message type: 127
Message before null: VISN 75210376
Message before that: ATTC 75210376
```

I couldn't spot anything in the logging code that points to a problem with these messages (formats). I don't know if Logmuncher reports the same errors but ignores them.

tumbili, 2015-08-13:
@AndreasAntener Seems like the logparser is not detecting the ATTC1 message. It is there in the log but he does not see it. I'm checking why right now.

tumbili, 2015-08-13:
@LorenzMeier @AndreasAntener Got it! Vision and ATTC1 have the same ID

tumbili, 2015-08-13:
We should have a script which checks that no ID is used two times.

LorenzMeier, 2015-08-17:
Fixed.

closed 2015-08-17 by LorenzMeier