PX4/PX4-Autopilot #2719: [TECS] Excessive pitch_sp when engaging TECS

boosfelm, 2015-08-18:
The issue of #2531 remains, sometimes when we go into Altitude Hold mode the ATSP_PitchSP goes to min or max value. Any idea what condition is hit by accident that it performs that way?
[image]
[image]

logs will be added as soon as the are uploaded.

LorenzMeier, 2015-08-18:
Thanks for reporting - I don't have a suspicion off the top of my head.

LorenzMeier, 2015-08-18:
Is there some way for you to repro this in HIL?

boosfelm, 2015-08-19:
here the log from the first screenshot:
http://dash.oznet.ch/view/g3JvqUdUiQpN86frZUa87i

Regarding HIL I am not sure, I will have to talk to the guys.

tumbili, 2015-08-19:
@boosfelm Are you sure this is the same log from the screenshot?

tumbili, 2015-08-19:
[image]

tumbili, 2015-08-19:
@boosfelm @LorenzMeier However, this log also shows interesting things. At the moment when you switch to altitude control mode the pitch setpoint goes up and from the airspeed reading you would assume the aircraft was stalling. This is probably also why you have switched out of this mode again. The altitude error at this point (not plotted) was very small, so I can't see a reason why the pitch should have gone up like crazy.

tumbili, 2015-08-19:
Notice also that the second try seemed to work well. Max, can you explain if my observation is consistent with you flight experience?

LorenzMeier, 2015-08-19:
You are just looking at the symptoms, not at the actual control action. You need to plot the TECS energy and energy allocation, in particular TER and EDR. These should start at zero on mode switching and then build up. The shared log is not plotting them correctly, here is an example from another log where they spike incorrectly:
http://dash.oznet.ch/view/MRjW8NUNYQSuSZkbn8dEjY#TECS_Inner_Loop_PLOT

LorenzMeier, 2015-08-19:
https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2_messages.h#L372-L375
and
https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2_messages.h#L548

Total energy rate = integ6_state:
https://github.com/PX4/Firmware/blob/master/src/lib/external_lgpl/tecs/tecs.cpp#L367

Energy distribution rate = integ7_state:
https://github.com/PX4/Firmware/blob/master/src/lib/external_lgpl/tecs/tecs.cpp#L454

Its pretty simple: TER is the combined altitude and speed energy. EDR is how the energy error is compensated. TERSP is the desired total energy and EDRSP is the desired distribution rate.

I have long advocated a TECS rewrite / clean new implementatoin (as in using human-readable variables, not changing any of the logic (yet)). But it has to be done cleanly (in copyright terms, so not copying the file), well tested and in BSD + Eigen so that we can base on it in the future.

tumbili, 2015-08-19:
@LorenzMeier TER is the combined altitude and speed energy RATE, no? I believe we just log the rates but not the energies.

tumbili, 2015-08-19:
@boosfelm I've flown ALTCTL, POSCTL, RTL, LOITER and AUTO today for a while. Couldn't reproduce this phenomena you mentioned. As I said, I think you shared the wrong log here. If you share the correct one I can look further into this.

boosfelm, 2015-08-20:
@tumbili Thanks for checking again.
1) It is the correct log 672s in my first plot corresponds to 88s in yours.
2) Regarding the symptoms I also agree, i.e. when we enter altitude control the Pitch_SP jumps up to max (we also experienced minimal value in another instance) and our att controller tries to follow.
3) However, I am still usnure what is causing it to happen. My guess would be some cornercase in the initialization.
[image]

LorenzMeier, 2015-08-22:
This here could be a potential fix:
#2743

LorenzMeier, 2015-08-23:
Fixed on all branches according to our HIL and flight testing.

closed 2015-08-23 by LorenzMeier

boosfelm, 2015-08-23:
thanks for the fix! We will fligth test it next week!