PX4/PX4-Autopilot #2726: Logged motor outputs of deactivated motors are not "0"

AndreasAntener, 2015-08-19:
@tumbili  I'm not sure if this is really an issue (we see no symptoms), the disabled motors during fixed-wing flight with a VTOL configuration don't seem to stay at the lowest level but change slightly. Would be good to know if this is a problem or not.

Firefly example from an old log (http://dash.oznet.ch/view/SxJPFGEGfW9dwVrSPuXLvb):
[image]

Or here (http://dash.oznet.ch/view/jy9DdoXST5zy3X4uoKPhs4):
[image]

SimonWilks, 2015-08-19:
Here is a plot that @deadhurricane shared on gitter: 

[image]

Looks like we have a leak on disabled motors ;)

tumbili, 2015-08-20:
@AndreasAntener @SimonWilks I think I know what the reason for the leak is:

effective_pwm[i] = control_value * (max_pwm[i] - min_pwm[i])/2 + (max_pwm[i] + min_pwm[i])/2;
/* last line of defense against invalid inputs */
if (effective_pwm[i] < min_pwm[i]) {
	effective_pwm[i] = min_pwm[i];
} else if (effective_pwm[i] > max_pwm[i]) {
	effective_pwm[i] = max_pwm[i];
}

This is the code which makes sure that the outputs satisfy the pmw limits. In the first line our control signal is mapped to the range [min_pwm, max_pwm]. So the values for the motors are going to move inside this very small range, (e.g. 900-950) but they are never going to be out of the range which is our requirement. So everything is fine.

tumbili, 2015-08-20:
BTW, you find the code here https://github.com/PX4/Firmware/blob/2a8402edb1313d8e70301b8045bf0b5132a4c7ff/src/modules/systemlib/pwm_limit/pwm_limit.c#L196

tumbili, 2015-08-21:
@AndreasAntener @SimonWilks Closing this as it doesn't pose an issue.

closed 2015-08-21 by tumbili

AndreasAntener, 2015-08-21:
:+1: thanks for having a look