PX4/PX4-Autopilot #2727: With perm_stab off, attitude setpoint during back transition with standard VTOL is 0

AndreasAntener, 2015-08-19:
This essentially means that during the back transition there is no stabilization and practically no manual control. If the user doesn't do anything, the VTOL will keep the rates is has and continue that movement during the back transition time until multirotor kicks in.

See first back transition here: http://dash.oznet.ch/view/bgJVKEGhAtkcwqsRd46KE7
[image]

AndreasAntener, 2015-08-19:
We need to consider the transition state here:
```
control_mode.flag_control_rates_enabled = (status.is_rotary_wing || status.vtol_fw_permanent_stab);
control_mode.flag_control_attitude_enabled = (status.is_rotary_wing || status.vtol_fw_permanent_stab);
```
From FW to MC we're in fixed wing mode, so nothing forces stabilization. But we should have stabilization through the transition in any case.

AndreasAntener, 2015-08-19:
We'd have the same issue with the Firefly, but we always flew with perm_stab on and the back transition is only 1 sec and not 3 or 4 as we tried here.

SimonWilks, 2015-08-19:
The Firefly actually cuts thrust during the back transition so it shouldn't make a difference either way. It's assumed that if you are flying fully manually you will fly it through right into multirotor mode. So as long as FW control surfaces still function all should be good.

But this brings up a good point. We should probably limit the back transition duration time a user can set for the Firefly to be no more the 2 secs.

AndreasAntener, 2015-08-19:
> It's assumed that if you are flying fully manually you will fly it through right into multirotor mode.

I don't agree with that statement: during transition, something automatic is happening, and the user should not be able to mess with it (maybe control it to a certain extent, but rolling the plane as I did today should not be possible ;)). On the front transition this cannot happen at the moment, so at least it should be consistent. Your PR will take care of it :)

AndreasAntener, 2015-08-19:
About control during the back transition: On the standard VTOL we also might have the wrong approach there. It is blending all controls linearly, but the plane has most likely still enough airspeed so we should give roll and maybe pitch completely to FW for the duration of the transition. This is actually the case for roll in the tiltrotor code.

SimonWilks, 2015-08-19:
No, my PR won't take care of it. There is no blending on the way back for the Firefly because the motors are off while rotating back. If the user is flying manually they won't notice till multirotor takes over. If you want to let go of sticks you should be flying fixed wing in stabilised, not manual otherwise you must actively fly.

SimonWilks, 2015-08-19:
Re standard back transition, yes, I think something better could be done. Let's try it out and see how it looks. It should be an acceptable first step...I hope.

AndreasAntener, 2015-10-26:
This was fixed a while a ago.

closed 2015-10-26 by AndreasAntener