PX4/PX4-Autopilot #2729: FW: plane goes into endless dive if airspeed sensor fails

tumbili, 2015-08-19:
I landed my fw in the grass which probably clogged my airspeed sensor. After re-takeoff the plane always dived towards the ground in ALTCTL mode. Looking at the log the airspeed sensor show values which are way too low for what I flew.

Log:
http://dash.oznet.ch/view/4ChvyQdeEWpBwaHErYNEni

AndreasAntener, 2015-08-19:
Ha! You found one more reason to change something about this speed/altitude weight for alt/posctl modes in fixed wing.
https://github.com/PX4/Firmware/issues/2030

AndreasAntener, 2015-08-19:
It's too bad we didn't yet work out the final approach to solving this, but I think the ideas are there in the issue I referenced.

tumbili, 2015-08-19:
@AndreasAntener Yes true, although this time the extra flavour of an airspeed sensor providing wrong data comes into play :-)

AndreasAntener, 2015-08-19:
Sure, my first comment was a little overhasty. It's just a pity that my fixed wing flight testing in this area dropped to the bottom of the todo list, and I haven't tried any of the ideas yet. So, I'm done spamming this issue :)

LorenzMeier, 2015-08-28:
It needs an airspeed failure detector.

LorenzMeier, 2015-11-17:
This is a log showing complete failure of the airspeed sensor. The airspeed sensor has been calibrated before and was indicating values when blowing into it on the ground.
Log: http://dash.oznet.ch/view/pFDAKzzFivKPvwXwqtwwGK

LorenzMeier, 2015-12-02:
Same as on previous release, closing.

closed 2015-12-02 by LorenzMeier