PX4/PX4-Autopilot #2734: Fix code style in remaining apps, enforce code style via Travis

LorenzMeier, 2015-08-21:
@dagar Would you be interested to have another stab at this?

It seems like astyle 2.5.1 fixed this. Which means we need a PPA in Travis or build it from source so we have a good version:

```diff
-       _mission_item({0}),
-       _waypoint_position_reached(false),
-       _waypoint_yaw_reached(false),
-       _time_first_inside_orbit(0),
-       _actuators{},
-       _actuator_pub(nullptr)
+       _mission_item( {0}),
+              _waypoint_position_reached(false),
+              _waypoint_yaw_reached(false),
+              _time_first_inside_orbit(0),
+              _actuators{},
+              _actuator_pub(nullptr)
```

Now when using the correct {} instead of ({0}), which is anyway wrong because it invokes the copy constructor, astyle does not mess up the initialiser list any more.

LorenzMeier, 2015-08-21:
I would propose to:

- Enforce code style on all changed files (files which show up when running ```git diff <target_branch> --stat | grep src/ | cut -f2 -d' '``` e.g. by using ```Tools/fix_code_style.sh `git diff master --stat | grep src/ | cut -f2 -d' '```` and then checking if any file changed)

LorenzMeier, 2015-08-21:
We have reached a good point to be able to digest a massive code change in general. The only modules to not touch right now are:

  * sdlog2
  * vtol controllers
  * uORB

The rest of the code could be formatted now file-by-file and then enforced to follow the code style.

dagar, 2015-08-21:
I can't find an astyle 2.5.1 ppa so I'll see how badly building it from source hurts the build time.

dagar, 2015-08-22:
See PR #2744

dagar, 2016-02-27:
Anything else you'd like to do here?

closed 2016-07-29 by LorenzMeier