PX4/PX4-Autopilot #2737: Frame size

Royalcheese, 2015-08-21:
Hello everyone,
I'm not sure where to ask the question , so I post it both on google group and here.
I've just build a new non linear estimator on the basis of the module attitude_estimator_efk. But I obtain this at the end :

error: the frame size of 1312 bytes is larger than 1024 bytes [-Werror=frame-larger-than=]

Is there anyway to make a larger frame, or do i have to improve to code to spare memory ? maybe as class rather than an application it cause problems ? I use many matrix calculus and i'm not sure I can improve a lot my code. 

Any advice would be nice !

Thanks in advance and best regards,
CP

LorenzMeier, 2015-08-21:
You can do that in the module.mk, and you can look at other apps (look at estimators) to see how to do that. When you do, evaluate the stack size your app is using and increase it in the task_spawn command.

closed 2015-08-21 by LorenzMeier

Royalcheese, 2015-08-21:
Thank you a lot ! I just found it btw ^^