PX4/PX4-Autopilot #2738: Policy for running UAVCAN dynamic node ID allocation and firmware update servers

pavel-kirienko, 2015-08-21:
UAVCAN is implemented in the form of two processes:

* Main process that controls actuators and collects sensor data. It starts at boot time automatically if `UAVCAN_ENABLE` is greater than zero. It is OK to have it running nonstop because of low memory requirements.
* Secondary process that runs the title logic - dynamic node ID allocator and firmware updater. It will be started automatically at boot if `UAVCAN_ENABLE` is greater than one. It would be *preferred* to keep it running all the time as well from the bus usability standpoint, but it is considered to take too much memory for that to be possible.

I expect that most peripheral nodes will require dynamic node ID allocation, therefore the secondary process must be running at least for a limited time after boot. There must be some logic in place that will start the secondary process after boot and then terminate it automatically. The open questions are:

* What would be the termination condition for the second process?
* If a UAVCAN node is connected to the bus when the secondary process has already been terminated, how is it going to be allocated?
* This is an embedded system, its memory needs are unlikely to change as long as its configuration doesn't change. Therefore, if there was enough memory to start the servers after boot up, there will be enough memory to keep it running forever. If the reasoning is correct, what is the point in terminating the second process?

LorenzMeier, 2015-08-22:
The right logic would be to terminate it on arming, as the system config will not change after that. The reason this would be an additional benefit is that the logging to flash requires RAM to buffer, so the freed memory can be traded between config and logging.

pavel-kirienko, 2015-08-22:
Shall we restart the secondary thread when the system gets disarmed?

LorenzMeier, 2015-08-22:
Yes. It will be very confusing for users else.

pavel-kirienko, 2015-08-22:
@all, is there anybody willing to undertake this, or should I implement it?

closed 2015-12-20 by pavel-kirienko