PX4/PX4-Autopilot #2755: SITL joystick interface with MAV_TYPE 19

devbharat, 2015-08-24:
On firmware commit 9e1713eb9f81a28dea7c025a9eb1944d489afd07

Two points to note:
- On make sitl_quad, auto-launching of rcS script in Firmware/posix-configs/SITL/init/rcS does not happen anymore (It used to happen a couple of weeks ago, not anymore. I thought it was a new feature, but apparently, its not.)

- Interfacing joystick with QGCS when running SITL does not work when the SITL is launched with MAV_TYPE 19, but works for MAV_TYPE 2.  (Tested in QGC v2.5.2 and 2.7.1)

The joystick is still recognized by QGCS, but on doing radio calibration, the analogue sticks also behave like 3 point buttons, although the linux joystick driver reads  the stick value correctly.

Steps to reproduce:
Console 1:

    java -Djava.ext.dirs= -cp lib/*:out/production/jmavsim.jar me.drton.jmavsim.Simulator -udp 127.0.0.1:14560

Console 2:

    make posix
    make sitl_quad

It prompts for launching modules. I input the startup script Firmware/posix-configs/SITL/init/rcS with the exception that MAV_TYPE is 19, not 2.

    uorb start
    simulator start -s
    param load
    param set MAV_TYPE 19
    param set MC_PITCHRATE_P 0.15
    param set MC_ROLLRATE_P 0.15
    param set MC_YAW_P 2.0
    param set MC_YAWRATE_P 0.35
    param set SYS_AUTOSTART 4010
    param set SYS_RESTART_TYPE 2
    param set COM_RC_IN_MODE 2
    dataman start
    param set CAL_GYRO0_ID 2293760
    param set CAL_ACC0_ID 1376256
    param set CAL_ACC1_ID 1310720
    param set CAL_MAG0_ID 196608
    param set CAL_GYRO0_XOFF 0.01
    param set CAL_ACC0_XOFF 0.01
    param set CAL_ACC0_YOFF -0.01
    param set CAL_ACC0_ZOFF 0.01
    param set CAL_ACC0_XSCALE 1.01
    param set CAL_ACC0_YSCALE 1.01
    param set CAL_ACC0_ZSCALE 1.01
    param set CAL_ACC1_XOFF 0.01
    param set CAL_MAG0_XOFF 0.01
    param set MPC_XY_P 0.4
    param set MPC_XY_VEL_P 0.2
    param set MPC_XY_VEL_D 0.005
    param set SENS_BOARD_ROT 0
    rgbled start
    tone_alarm start
    gyrosim start
    accelsim start
    barosim start
    adcsim start
    gpssim start
    hil mode_pwm
    commander start
    sensors start
    land_detector start multicopter
    navigator start
    attitude_estimator_q start
    position_estimator_inav start
    mc_pos_control start
    mc_att_control start
    mixer load /dev/pwm_output0 ../../ROMFS/px4fmu_common/mixers/quad_x.main.mix
    mavlink start -u 14556 -r 2000000
    mavlink stream -r 80 -s POSITION_TARGET_LOCAL_NED -u 14556
    mavlink stream -r 80 -s LOCAL_POSITION_NED -u 14556
    mavlink stream -r 80 -s GLOBAL_POSITION_INT -u 14556
    mavlink stream -r 80 -s ATTITUDE -u 14556
    mavlink stream -r 80 -s ATTITUDE_TARGET -u 14556
    mavlink stream -r 20 -s RC_CHANNELS -u 14556
    mavlink stream -r 250 -s HIGHRES_IMU -u 14556
    mavlink boot_complete

Console 3:
Connect joystick to PC and Launch QGCS and connect to UDP. Mavlink connects. Go to radio calibration. Move roll/pitch/yaw sticks, I see that rather than going smoothly from-1 to +1 they jump from -1 to 0 to +1. This is the issue. It is not seen if the whole procedure is repeated with MAV_TYPE 2

devbharat, 2015-08-24:
It seems like QGCS joystick mapping does depend on the system and autopilot type:
/src/input/Joystick.h

    // Track axis/button settings based on a Joystick/AutopilotType/SystemType triplet.
    // This is only a double-map, because settings are stored/loaded based on joystick
    // name first, so only the settings for the current joystick need to be stored at any given time.
    // Pointers are kept to the various settings field to reduce lookup times.
    // Note that the mapping (0,0) corresponds to when no UAS is connected. Since this corresponds
    // to a generic vehicle type and a generic autopilot, this is a pretty safe default.
    QMap<int, QMap<int, JoystickSettings> > joystickSettings;

devbharat, 2015-08-24:
And a little hack to force systemType to 2 makes it work. Of course, there must be a better way of doing it.

void JoystickInput::setActiveUAS(UASInterface* uas)
    {
        // Do nothing if the UAS hasn't changed.
        if (uas == this->uas)
        {
            return;
        }

        // Only connect / disconnect is the UAS is of a controllable UAS class
        UAS* tmp = 0;
        if (this->uas)
        {
            tmp = dynamic_cast<UAS*>(this->uas);
            if(tmp)
            {
                disconnect(this, SIGNAL(joystickChanged(float,float,float,float,qint8,qint8,quint16,quint8)), tmp, SLOT(setExternalControlSetpoint(float,float,float,float,qint8,qint8,quint16,quint8)));
                disconnect(this, SIGNAL(actionTriggered(int)), tmp, SLOT(triggerAction(int)));
            }
            uasCanReverse = false;
        }

        // Save any settings for the last UAS
        if (joystickID > -1)
        {
            storeJoystickSettings();
        }

        this->uas = uas;

        if (this->uas && (tmp = dynamic_cast<UAS*>(this->uas)))
        {
            connect(this, SIGNAL(joystickChanged(float,float,float,float,qint8,qint8,quint16,quint8)), tmp, SLOT(setExternalControlSetpoint(float,float,float,float,qint8,qint8,quint16,quint8)));
            qDebug() << "connected joystick";
            connect(this, SIGNAL(actionTriggered(int)), tmp, SLOT(triggerAction(int)));
            uasCanReverse = tmp->systemCanReverse();

            // Update the joystick settings for a new UAS.
            autopilotType = uas->getAutopilotType();
            systemType = 2;//uas->getSystemType(); // HACK HACK HACK
        }

        // Make sure any UI elements know we've updated the UAS. The UASManager signal is re-emitted here so that UI elements know to
        // update their UAS-specific UI.
        emit activeUASSet(uas);

        // Load any joystick-specific settings now that the UAS has changed.
        if (joystickID > -1)
        {
            loadJoystickSettings();
        }
    }

DonLakeFlyer, 2015-09-01:
I think since Joystick settings are stored keyed by MAV_AUTOPILOT and MAV_TYPE you need to have calibrated with that type a vehicle activee. If you calibrated with a different vehicle it won't pick that one up. So, although the current implementation seems odd, it is working correctly.

closed 2015-12-10 by LorenzMeier