PX4/PX4-Autopilot #2756: Unwanted yawing after transition from FW to MC with Tiltrotor in ALTCTL

AndreasAntener, 2015-08-24:
@tumbili 
Firefly did a 360 after the transition. Fix for yaw setpoint update during the transition was already in. Initial thoughts on this were about possible mixer saturation.

I'm not sure anymore but I think this also happened in this log: http://dash.oznet.ch/view/jL3B7crHoTEmveAWywHW8g

tumbili, 2015-08-24:
@AndreasAntener @SimonWilks @LorenzMeier To me it looks pretty obvious. When the multicopter motors kick in the plane starts to increase altitude (vz was -3.7 m/s). In order to prevent this the mc position controller lowered the thrust down to 14.2 %. At the same time we had a relatively high ATTC.Roll and the mixer went into saturation (actuator outputs reached lower limits). Together with the fact that yaw actuation is poor for the FireFly this makes perfect sense to me.

tumbili, 2015-08-24:
And by the way, airspeed at this point was still above 10 m/s so the multicopter position controller tried to control a plane thinking it was a multicopter.

AndreasAntener, 2016-01-11:
@tumbili did we fix this one?

tumbili, 2016-02-14:
Fixed!

closed 2016-02-14 by tumbili