PX4/PX4-Autopilot #2757: Current master is not working with PX4FMU_v1

limhyon, 2015-08-25:
Hi,
I downloaded current master to my PX4FMU_V1, but it is not working.
It is frozen during boot. I attached log file as follows.

[Start of Log]

[boot] Enabling IN12/13 instead of SPI2
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
nsh: mount: mount failed: No such device
nsh: mkfatfs: mkfatfs failed: I/O error
nshterm [6:70]
param: selected parameter default file /fs/mtd_params
[param] Loaded: /fs/mtd_params
 0:   RC_MAP_THROTTLE: match
 1000.0000000:   RC3_MIN: match
 0:   RC_MAP_ROLL: match
 1000.0000000:   RC1_MIN: match
 0:   RC_MAP_PITCH: match
 1000.0000000:   RC2_MIN: match
 0:   RC_MAP_YAW: match
 1000.0000000:   RC4_MIN: match
rgbled: no RGB led on bus #3
blinkm: I2C init failed
blinkm: init failed
 1:   SYS_USE_IO: match
 0:   SYS_AUTOSTART: match
[i] No autostart
PX4IO_i2c on I2C bus 2 at 0x1a (bus: 100 KHz, max: 400 KHz)
px4io: check CRC failed: -1: Invalid argument
PX4IO_i2c on I2C bus 2 at 0x1a (bus: 100 KHz, max: 400 KHz)
telem> Failed to communicate with IO, abort.
px4io: driver init failed
px4io: px4io is not started, still attempting upgrade
PX4IO_i2c on I2C bus 2 at 0x1a (bus: 100 KHz, max: 400 KHz)
[PX4IO] using firmware from /etc/extras/px4io-v1_default.bin
[PX4IO] bootloader not responding
PX4IO_i2c on I2C bus 2 at 0x1a (bus: 100 KHz, max: 400 KHz)
px4io: check CRC failed: -1: Invalid argument
[i] ERROR: PX4IO not found
[i] ERROR: PX4IO not found, disabling output
ver hwcmp match: PX4FMU_V1
dataman: Could not open data manager file /fs/microsd/dataman
dataman: dataman start failed
param: Error: Parameter UAVCAN_ENABLE not found.
param: Error: Parameter UAVCAN_ENABLE not found.
[i] UAVCAN_ENABLE is 0
MS5611_I2C on I2C bus 2 at 0x76 (bus: 100 KHz, max: 400 KHz)
ms5611: no device on bus 2
ms5611: driver start failed
adc init done
MPU6000 on SPI bus 1 at 3 (1000 KHz)
L3GD20 on SPI bus 1 at 1 (11000 KHz)
hmc5883: no device on bus 2
HMC5883_I2C on I2C bus 2 at 0x1e (bus: 100 KHz, max: 400 KHz)
meas_airspeed: no MS4525 airspeed sensor connected
ets_airspeed: no ETS air

[End of Log]

LorenzMeier, 2015-08-25:
Please re-run make distclean && make archives and rebuild - let us know if the issue persists then.

LorenzMeier, 2015-08-25:
Note that your microSD is no good:
nsh: mount: mount failed: No such device

LorenzMeier, 2015-11-17:
Closing this, as FMUv1 is tested.

closed 2015-11-17 by LorenzMeier