PX4/PX4-Autopilot #2763: Current master doesn't boot on Pixhawk

AndreasAntener, 2015-08-26:
Tried on different Pixhawks, I don't seem to find the error. beta and stable work.
Forced IO update, tried flashing from QGC and a clean build on my machine.

Maybe related to #2757 

```
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] microSD mounted: /fs/microsd 
nshterm [6:70]
param: selected parameter default file /fs/mtd_params
[param] Loaded: /fs/mtd_params 
[rgbled] on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
 1:   SYS_USE_IO: match
 4010: + SYS_AUTOSTART: match
[i] Custom config: /fs/microsd/etc/config.txt 
px4io: check CRC failed: -22: Invalid argument
[px4io] default PWM output device
[PX4IO] using firmware from /etc/extras/px4io-v2_default.bin
[PX4IO] found bootloader revision: 4
[PX4IO] erase...
[PX4IO] programming 47360 bytes...
[PX4IO] verify...
[PX4IO] update complete
px4io: CRCs match
dataman: Power on restart, data manager file '/fs/microsd/dataman' size is 101504 bytes
[i] UAVCAN_ENABLE is 0 
[MS5611_SPI] on SPI bus 4 at 2 (11000 KHz)
ms5611: no device on bus 4
[MS5611_SPI] on SPI bus 1 at 3 (11000 KHz)
ms5611: no device on bus 1
ms5611: no device on bus 2
<adc> init done
ver hwcmp match: PX4FMU_V2
[HMC5883_I2C] on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
hmc5883: no device on bus 1
mpu6000: driver start failed
[MPU6000] on SPI bus 1 at 4 (1000 KHz)
[L3GD20] on SPI bus 1 at 1 (11000 KHz)
[LSM303D] on SPI bus 1 at 2 (11000 KHz)
meas_airspeed: no MS4525 airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
sensors_task: mag cal status changed 
[px4io] default PWM output device
mavlink_if0: mode: 0, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
ver hwcmp match: PX4FMU_V2
 921600: + SYS_COMPANION: match
mavlink_if1: mode: 2, data rate: 80000 B/s on /dev/ttyS2 @ 921600B
sdlog2: log buffer size: 12288 bytes
[i] MULTICOPTER 
+ MAV_TYPE: unchanged
[px4io] mixer sent
[i] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0 
[i] Mixer: /fs/microsd/etc/mixers/gimbal.aux.mix on /dev/pwm_output1 
 1:   INAV_ENABLED: match
param: Error: Parameter EKF_ATT_ENABLED not found.
attitude_estimator_q: started
[i] Addons script: /fs/microsd/etc/extras.txt 
ver hwcmp match: PX4FMU_V2
px4flow [188:100]

NuttShell (NSH)
nsh> px4flow: scanning I2C buses for device..
mavlink_if1: Disabling hardware flow control
gps: module found: UBX
telem> settings autosaved
mavlink_if0: Disabling hardware flow control
position_estimator_inav: baro offset: 309 m
```

AndreasAntener, 2015-08-26:
Symptoms:
- IO B/E light blinks
- main LED just lights up for a brief blink (red), then stays dark

AndreasAntener, 2015-08-26:
Output from a more recent board:
```
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] microSD mounted: /fs/microsd 
nshterm [6:70]
param: selected parameter default file /fs/mtd_params
[param] Loaded: /fs/mtd_params 
 0:   RC_MAP_YAW: match
+ RC_MAP_ROLL: unchanged
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
 1:   SYS_USE_IO: match
 4010: + SYS_AUTOSTART: match
[i] Custom config: /fs/microsd/etc/config.txt 
px4io: CRCs match
dataman: Power on restart, data manager file '/fs/microsd/dataman' size is 101504 bytes
[i] UAVCAN_ENABLE is 0 
MS5611_SPI on SPI bus 4 at 2 (11000 KHz)
ms5611: no device on bus 4
MS5611_SPI on SPI bus 1 at 3 (11000 KHz)
ms5611: no device on bus 1
ms5611: no device on bus 2
adc init done
ver hwcmp match: PX4FMU_V2
HMC5883_I2C on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
hmc5883: no device on bus 1
mpu6000: driver start failed
MPU6000 on SPI bus 1 at 4 (1000 KHz)
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
meas_airspeed: no MS4525 airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
sensors_task: mag cal status changed 
px4io default PWM output device
mavlink_if0: mode: 0, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
ver hwcmp match: PX4FMU_V2
 921600: + SYS_COMPANION: match
mavlink_if1: mode: 2, data rate: 80000 B/s on /dev/ttyS2 @ 921600B
sdlog2: log buffer size: 12288 bytes
[i] MULTICOPTER 
+ MAV_TYPE: unchanged
px4io mixer sent
[i] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0 
[i] Mixer: /fs/microsd/etc/mixers/gimbal.aux.mix on /dev/pwm_output1 
 1:   INAV_ENABLED: match
param: Error: Parameter EKF_ATT_ENABLED not found.
[i] Addons script: /fs/microsd/etc/extras.txt 
ver hwcmp match: PX4FMU_V2
px4flow [190:100]

NuttShell (NSH)
nsh> px4flow: scanning I2C buses for device..
mavlink_if1: Disabling hardware flow control
gps: module found: UBX
mavlink_if0: Disabling hardware flow control
position_estimator_inav: baro offset: 310 m

```

LorenzMeier, 2015-08-26:
Try rebuilding the archives (or flash via QGC) and let us know what happens. If that doesn't help, disable the PX4flow start in the rcS (wild guess).

LorenzMeier, 2015-08-26:
And I presume this was before your RC_MAP fix, as it looks wrong?
```
 0:   RC_MAP_YAW: match
+ RC_MAP_ROLL: unchanged
```

AndreasAntener, 2015-08-26:
Yest that was a try before my fix. Also tried flashing from QGC already, same issue.
Disabled px4flow, same issue:
```
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] microSD mounted: /fs/microsd 
nshterm [6:70]
param: selected parameter default file /fs/mtd_params
[param] Loaded: /fs/mtd_params 
 0:   RC_MAP_YAW: match
  RC_MAP_YAW: curr: 0 -> new: 4
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
 1:   SYS_USE_IO: match
 4010: + SYS_AUTOSTART: match
[i] Custom config: /fs/microsd/etc/config.txt 
px4io: CRCs match
dataman: Power on restart, data manager file '/fs/microsd/dataman' size is 101504 bytes
[i] UAVCAN_ENABLE is 0 
MS5611_SPI on SPI bus 4 at 2 (11000 KHz)
ms5611: no device on bus 4
MS5611_SPI on SPI bus 1 at 3 (11000 KHz)
ms5611: no device on bus 1
ms5611: no device on bus 2
adc init done
ver hwcmp match: PX4FMU_V2
HMC5883_I2C on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
hmc5883: no device on bus 1
mpu6000: driver start failed
MPU6000 on SPI bus 1 at 4 (1000 KHz)
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
meas_airspeed: no MS4525 airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
sensors_task: mag cal status changed 
px4io default PWM output device
mavlink_if0: mode: 0, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
mavlink_if0: offboard mission init: ERROR
ver hwcmp match: PX4FMU_V2
 921600: + SYS_COMPANION: match
mavlink_if1: mode: 2, data rate: 80000 B/s on /dev/ttyS2 @ 921600B
sdlog2: log buffer size: 12288 bytes
[i] MULTICOPTER 
+ MAV_TYPE: unchanged
px4io mixer sent
[i] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0 
[i] Mixer: /fs/microsd/etc/mixers/gimbal.aux.mix on /dev/pwm_output1 
 1:   INAV_ENABLED: match
param: Error: Parameter EKF_ATT_ENABLED not found.
[i] Addons script: /fs/microsd/etc/extras.txt 
ver hwcmp match: PX4FMU_V2

NuttShell (NSH)
nsh> mavlink_if1: Disabling hardware flow control
gps: module found: UBX
mavlink_if0: Disabling hardware flow control
position_estimator_inav: baro offset: 316 m
```

AndreasAntener, 2015-08-26:
Seems 3a474347491236fcb32ac6b58fe7d0cf78564f3b broke it

LorenzMeier, 2015-08-26:
@mcharleb The offending code piece is in fact this. There are two issues:

1) The argc logic didn't work
2) PX4_ERR does not print anything in NuttX

```
	if (argc > 0) {
		if (strcmp(argv[0],"-hil")==0) {
			startup_in_hil = true;
		} else {
			PX4_ERR("Argument %s not supported.",argv[0]);
			PX4_ERR("COMMANDER NOT STARTED");
			thread_should_exit = true;
			warnx("ERR");
		}
	}
```

LorenzMeier, 2015-08-26:
@AndreasAntener I fixed this for good in 9ef9185. @mcharleb I've renamed this issue and kept it open so we can fix the remaining PX4_ERR call to print something on NuttX.

LorenzMeier, 2015-10-14:
Fixed.

closed 2015-10-14 by LorenzMeier