PX4/PX4-Autopilot #2768: Altitude control: Energy reset on initial switch to althold not working

LorenzMeier, 2015-08-26:
On switching initially to altitude hold the energy integral is non-zero.

boosfelm, 2015-08-28:
This bug did not occur for us again. We had 4 flights yesterday and initialization worked just fine. We integrated #2743 #2747 and #2731 beforehand. Apart from that very good experience with the PX4 flight stack again.

tumbili, 2015-08-28:
@boosfelm Could you share the log please?

boosfelm, 2015-08-28:
sure, http://dash.oznet.ch/view/aLG7763JnYqDLapH5VaDqV

tumbili, 2015-08-28:
@boosfelm Thanks! Makes sense, TECS.PtchI starts at zero as how it's supposed to.

boosfelm, 2015-08-28:
here the second. they all look the same. http://dash.oznet.ch/view/6eUxb6ueVRqK4ZX6s5DL2W
Just weird why it did happen to you again after you applied the fix. maybe you did not reflash ;)

tumbili, 2015-08-28:
@LorenzMeier @boosfelm This is a log showing the problem with latest beta. We flashed the autopilot once and never again. Sometimes the integrator starts at zero sometimes it starts at a crazy value.

Log: http://dash.oznet.ch/view/BWu8DtL5tQgzFyB8ZR4xNY

LorenzMeier, 2015-08-28:
@boosfelm This should cover the last possible corner cases.
https://github.com/PX4/Firmware/pull/2783

boosfelm, 2015-08-30:
Cool, thanks a lot! I will report back if anything unexpected happens, but the fix looks really good to me!

LorenzMeier, 2015-08-30:
I flew this in HIL and in reality today, no glitches. Will provide the log in a sec.

LorenzMeier, 2015-08-30:
http://logs.uaventure.com/view/WGXFGt66aN4bNKjNKa7RZ3

closed 2015-08-30 by LorenzMeier