PX4/PX4-Autopilot #2772: STAB mode in V1.1.0 / firmware error?

severinleuenberger, 2015-08-27:
I tested my setup against px4 firmware v1.1.0 with my fixed wing "big turtle". I have flown this airplane since 14 month without any problems on stablized mode in various firmware developement steps from px4 (I did all the laser landing testing with ZHAW with this aircraft). 

On normal flight behaviours (smooth movements on roll and pitch), everything worked fine on my recent test of v1.1.0. So there was everything well. 
But I wanted to "rock" the plane respectivly the flight controller and dit some fast movements on roll. Even this was going ok. But in my last few flight seconds, I combined fast rolls with very narrow turns, and then it happended. The plane was going in a fast vertical spin until impact. It was to fast for me to switch to "manual" and got the following result...
[image]
Ok. Bad luck, fatal damage.  

But I would be very pleased, if you could analyze if there was a px4-firmware funcitonal issue behind. Of course, it could be any technical issue other than the firmware. 

log muncher is here:
http://dash.oznet.ch/view/okTSrSHWyK4sjSCbqvP8Cn
Thanks for analyzing! I cannot see what could be the cause here. 

Best wishes
Severin

severinleuenberger, 2015-08-27:
Comment: I dint overstressed the airframe. It was a very narrow turn, but not aerodynamically critical. 
I have been flying the airframe since 2.5 years and it was everything in borders of "not critical" in point of view from the aircraft itself. And I have done the same maneuvers with the flight controllers APM and FY-DOS before, and it was going well.

AndreasAntener, 2015-08-27:
When exactly was it out of control for you? Towards the end you were in STAB for about 1 min (~from 460 to 520). At the time you switched to STAB it went into a fast descent already. Before it hit the ground you gave full rudder (~500). Was this the attempt to stop the spiral?

Could you also post the contents of the preflight and postflight txt files for this log?

severinleuenberger, 2015-08-27:
I would say out of control was on 432.0 and impact on 434.40. 
Maybe you can provide me a flightplot preset to monitor my RC inputs on roll, pitch, throttle and yaw? 

[image]

AndreasAntener, 2015-08-28:
One of us is looking at the wrong log then. That's what I have at the end from the log you posted in this issue:
[image]

LorenzMeier, 2015-08-28:
Using the log muncher flight data I can see a clear pilot-induced stall.

[image]

closed 2015-08-28 by LorenzMeier

severinleuenberger, 2015-08-28:
I am very sorry Lorenz, I uploaded the wrong file. .... 
Your analysis references to a nice and good working high alpha landing from my previous flight. 
To not generate to much confusion, I do open a new issue with the correct file!
Best, Severin

tumbili, 2015-08-28:
@severinleuenberger @LorenzMeier @AndreasAntener The lession that I learnt from this is that a huge spike in the accelerometer does not necessarily mean crash. Obviously also nice belly landing cause these accelerations. However, if you look at the gps speed it's apparent that this was a nice landing.

severinleuenberger, 2015-08-28:
@tumbili  where do you see a huge spike in the accelerometer in the above chart? Just for me to learn to interpret logs. 
And of course, the big turtle is somehow a littlebit a strange aircraft. 
Just my 2 cent about "stabilized mode" would be, that it mostly refers to the gyro and accelerometers, and the scaling factor from "airspeed" should only be used, if there is oscillation from to big servo movements. Otherwise its more safe to use stabilized most simple to just level the plane. 
Even baro should not be used in stabilized from my point of view. Just leveling the plane and allowing the max bank and pitch angles als defined in the param-list.  But i will reopen this disscussion in the new issue. I do not have access to the sd card now, this evening i can do this.

tumbili, 2015-08-28:
@severinleuenberger Check the "raw acceleration" chart on logmuncher.