PX4/PX4-Autopilot #2777: Fixed wing attitude stabilization becomes unstable with GPS issues

AndreasAntener, 2015-08-28:
@severinleuenberger: reporting your issue with GPS in stabilized here. @tumbili  as discussed.
Log: http://dash.oznet.ch/view/LgdXyaqoAPrzWtGv94VMHk

According to Severing the plane became unstable in stabilized after which he switched out to manual where it was flying normally.

[image]

It seems when GPS eph/epv goes beyond 10 the problem started. This should make the EKF estimator stop usign GPS: https://github.com/PX4/Firmware/blob/master/src/modules/ekf_att_pos_estimator/ekf_att_pos_estimator_main.cpp#L1376 . `requiredAccuracy` is doubled after the first fix (from 5 to 10)

severinleuenberger, 2015-08-28:
I have to say, that I mounted an old cheap GPS on that day in my big turlte fixed wing airplane. And not having a good GPS seems to lead to a very shaky and unstable flight performance in althold. My gut feeling would say, that in ALTHOLD mode the scaling factor of the GPS against the other sensors are overweightened. Better trust more to the baro and gyro / acc / airspeed sensor, and leave in case of uncertainity the GPS out of the calculation. Just my 2 cent based on some assumptions. 

One day later, where I mounted an expensive goog GPS again, the aircraft was flying very nicely again.

LorenzMeier, 2015-08-28:
Its not that simple. You will crash if you continue to fly circles in one direction when you don't use GPS for attitude estimation. Fundamentally for all robots its garbage in = garbage out.

severinleuenberger, 2015-08-28:
Circles would be ok, because that is not fundamentally a critical flight situation. But I had problem with pitch there as well, the plane was flying waves like on high waves on water. If the fmu alanyzes bad gps health, it should basically switch to STABILIZED mode. Just a suggestion.

AndreasAntener, 2015-08-29:
Severin, the problem is that IF you fly circles continuously, you need GPS to correct attitude.

@LorenzMeier The estimators should at least not get into bad state if GPS fails (as long as you don't fly circles). I'm a little worried else that a multirotor will fall out of the sky when it flies underneath a tree.

AndreasAntener, 2015-10-22:
@tumbili @LorenzMeier Something to consider in the current EKF iteration? I think this is still an issue.

LorenzMeier, 2015-10-22:
We will look at this with the new estimator. Not before.

tumbili, 2015-10-22:
@AndreasAntener @LorenzMeier Just did a test with SITL. Everything working smooth with the q estimator and inav, whereas the ekf just goes totally crazy. Complains all the time about excessive gyro offsets.

tumbili, 2015-10-22:
Tried using the ekf with SITL for multirotors, there it segfaults.

AndreasAntener, 2015-10-22:
@tumbili thanks already for testing :)

tumbili, 2016-02-14:
Switching to ekf2!

closed 2016-02-14 by tumbili