PX4/PX4-Autopilot #2784: STAB mode in V1.1.0 / cause of crash?

severinleuenberger, 2015-08-28:
Here we go again - ths time with the corect link to log muncher
******************************************************************************
I tested my setup against px4 firmware v1.1.0 with my fixed wing "big turtle". I have flown this airplane since 14 month without any problems on stablized mode in various firmware developement steps from px4 (I did all the laser landing testing with ZHAW with this aircraft).

On normal flight behaviours (smooth movements on roll and pitch), everything worked fine on my recent test of v1.1.0. So there was everything well. 
But I wanted to "rock" the plane respectivly the flight controller and dit some fast movements on roll. Even this was going ok. But in my last few flight seconds, I combined fast rolls with very narrow turns, and then it happended. The plane was going in a fast vertical spin until impact. It was to fast for me to switch to "manual" and it ended in a crash. Bad luck. 

But I would be very pleased, if you could analyze if there was a px4-firmware funcitonal issue behind. Of course, it could be any technical issue other than the firmware.

correct log muncher is here:
http://dash.oznet.ch/view/qqPccTguR2kGevZb5A8uKo

When I analyze the log, and compare it with my memories of the fight, the problems begun at 431.00 where the airplane begun to loose height and there was a major ssue with pitch at 431.30, until impact at 434.40

[image]

Thanks for analyzing! I have my gut feeling that it could be something with the logic in PX4 STAB-mode, but I do not understand the code there, so maybe you can have a look at.  

Best wishes
Severin

severinleuenberger, 2015-08-28:
There are 2 flights on this log. The first was smooth and nice as anything, a very good and stable flight performance. I refer to the second flight on this log, where I did some fast roll movemets and quick turns.

LorenzMeier, 2015-08-28:
Looks like some sort of aerodynamic or tuning issue on steep bank angles. The commanded pitch angle was a different one and the error was sensed correctly. I can also see a clear response from the actuators.

Your pitch tracking is also not very good for the rest of the flight, the most likely reason is a too soft pitch tuning which makes pitch not fully controllable.
http://dash.oznet.ch/view/qqPccTguR2kGevZb5A8uKo#AS_Pitch_PLOT

There is nothing in the log suggesting a system failure and this should be preventable with more careful pitch tuning and performance assessment.

closed 2015-08-28 by LorenzMeier