PX4/PX4-Autopilot #2790: Altitude hold performance for multicopters with default gains

LorenzMeier, 2015-08-30:
@SimonWilks @AndreasAntener Could you share your altitude control tuning gains? The ones I tried were not great:
http://logs.uaventure.com/view/xd3G45GPty2yZLwuaiXJB

AndreasAntener, 2015-09-01:
@LorenzMeier We were looking through some old flights with the F450 today. There was one day where we tried tuning Z with @tumbili:
```
1	50	MPC_Z_FF	0.5	9
1	50	MPC_Z_P	1.2	9
1	50	MPC_Z_VEL_D	0.001	9
1	50	MPC_Z_VEL_I	0.04	9
1	50	MPC_Z_VEL_MAX	2	9
1	50	MPC_Z_VEL_P	0.5	9
```
But I think we had VEL_D higher as well but I don't seem to find those flights anymore. We ended up reducing VEL_P again to 0.3 because 0.5 is pretty aggressive.

tumbili, 2015-09-02:
@LorenzMeier @AndreasAntener @SimonWilks On the dev call we were also talking about poor altitude hold performance in hight winds or when flying at higher velocities. Would be interesting to see if we can shield the autopilot better in order to reduce this effect. I find it quite annoying to be honest.

SimonWilks, 2015-09-02:
@LorenzMeier @tumbili I should point out that we only use INAV. It might be time to give EKF another go.

It would be good if we could measure this effect better/more accurately. It would help us more accurately quantify our testing results. So far we can only eyeball it and compare it to GPS but using laser might be a good alternative.

It's of course going to be near impossible to remove the variances when using only using a baro due to the vacuum that builds around the vehicle when it moves through the air but there is probably room for improvement. I'd also like to redo a side-by-side test I did with an F450/Pixhawk and a commercial (Blade) quadcopter but my initial impression was that the Blade did better, though it was inside a body. @tumbili what's the Iris like?

LorenzMeier, 2015-09-02:
Controlling the airflow around the baro is absolutely critical - for either filter. And you might need to increase the Z velocity weight for GPS. That all said assuming that its an observation problem. Testing with the laser on the football field would be the best you can do. @tumbili From the system performance point of view altitude hold is a priority, because its the weakest point of our multicopter codebase. Everything else looks really good.

LorenzMeier, 2015-09-02:
As you can see here the filter did observe the altitude loss correctly, but the controller didn't fight it sufficiently.
[image]

LorenzMeier, 2015-09-02:
The data is suggesting that the system is under-estimating the vertical velocity if GPS were right. That means that increasing MPC_Z_FF is the most likely candidate to improve the control response, assuming that I'm guessing right that its a direct feedforward from altitude error, rather than going through the nested altitude rate loop like the MPC_Z_P value does.

[image]

LorenzMeier, 2015-09-02:
Plus of course increasing the GPS velocity weight, but that works only well on ublox 7 and above (which I happen to be using).

AndreasAntener, 2015-09-02:
We just did some flights with EKF and a Lidar: http://logs.uaventure.com/view/Di4Pr5TMwRqKTPsuMRaVve

[image]

AndreasAntener, 2015-09-02:
The screenshot is from one cycle forwards/backwards. It was a slow motion to keep pitch small to not screw up the laser measurement (attitude in the screenshot is scaled by 10).

AndreasAntener, 2015-09-09:
@LorenzMeier after hacking the CAN GPS and another mag together ;) here's the result with INAV: http://logs.uaventure.com/view/kbxhPy4ufTY8bBKFJfczY7

During the flight we changed the following parameters:
```
Z_FF from 0.5 to 1
INAV_W_Z_GPS_P from 0.005 to 0.05
INAV_W_Z_GPS_V from 0 to 1
Z_VEL_D from 0 to 0.003
```

We saw no change in its behaviour at all.

These are 2 examples of the many forwards/backwards flights:
[image]

It is always the same pattern. Baro suddenly shows a big raise, where as GPS and laser don't change much. GPOS alt just goes after the baro and commands it to sink. The massive drop is clearly shown by GPS and laser.

Observations:
- GPS and laser follow each other pretty close
- It seems that changing the GPS weights for INAV had no effect. Ideas?
- Even if GPOS would be accurate, it is also clearly overshooting in both directions. Ideas? Do we need much higher D values?

AndreasAntener, 2015-09-14:
I'd like to share another interesting find with INAV: In this "longer" flight we saw a constant altitude increase over the whole flight (it was clearly visible because we had buildings as reference). GPS shows it, baro shows it, but GPOS altitude ignores that completely and doesn't show the slow height changes. In the middle of the flight you see a commanded height increase and decrease, where everything seems to behave normally.

[image]

LorenzMeier, 2015-09-14:
Have you looked at the exact ACC Z mean? What is it?

tumbili, 2015-09-14:
@AndreasAntener Could you also share the log of the second flight please?

AndreasAntener, 2015-09-14:
http://logs.uaventure.com/view/ztnZALTtwiW4MwqvesrVY8

v01d, 2015-09-14:
Hi, I'd like to share the link to a discussion from last week related to this problem, which mostly affects altitude control based on sonar: https://groups.google.com/d/msg/px4users/eMdM2jC47TQ/AYIGui1kAgAJ
Another user also somehow confirmed a similar problem.

This is with INAV (since EKF does not still provide PX4FLOW position control AFAIK, which I need) so I'm not sure if it applies to your discussion. However, I suspect EKF would be able to deal with this much more robustly.

ilia-sheremet, 2015-09-14:
I am another that another user =) And I would like to confirm that sonar data is not taken into account for local position estimation without GPS.

LorenzMeier, 2015-11-10:
@kd0aij Based on your experience we essentially just need to use units with proper cases?

kd0aij, 2015-11-10:
I think you need both a good case (to block light and filter pressure noise) and a good canopy to reduce aerodynamic effects. @LorenzMeier, @tumbili  and @SimonWilks have already mentioned above that air currents influence the barometric pressure reading.

I think we we'd be better off using a pressure sensor with a port which can be connected to a proper static pressure probe. Without such a probe, the entire structure containing the baro sensor becomes the "static probe". I noted in flight testing for #3150 that my QAV250/X2 baro reading is much more sensitive to lateral motion than longitudinal. I suspect that is due to better shielding by the frame when flying forward or backward. I'll do a test flight with the frame sides covered to see if that helps lateral performance.

LorenzMeier, 2015-11-10:
Yes, a static port seems to be a given.

kd0aij, 2015-11-10:
Just did some more ALTCTL testing with the QAV250. Here are my observations:

with left side shielded, lost altitude with lateral motion both left and right, about 1 meter
Covering both sides did not decrease the altitude loss, it's possibly even greater now. 
Since loss of altitude indicates that baro pressure is decreasing (making the autopilot think the vehicle is climbing), lateral motion is causing a decrease of air pressure at the barometer.

This indicates that it isn't airflow pressurizing the fuselage. Since longitudinal motion does not have nearly as large an effect as lateral, perhaps lateral motion is translating the baro into rarified air, a low-pressure region created by the downflow of air through the propeller discs. The rotor geometry of the QAV250 would support this hypothesis.

We can test that hypothesis by flying different geometries; my AeroQuad/X2 has 10" props on an X 650mm frame, so perhaps it will demonstrate the effect only on diagonal translations. I'll see if I can get it into the air soon.

LorenzMeier, 2015-11-10:
Mark, it should be possible to come up with a baro cover for the X2 allowing you to put a static port in various locations. Would that be a worthwhile Makerbot job?

kd0aij, 2015-11-10:
I'm afraid that's a bit beyond my skill level with OpenSCAD, but I think it should be possible with some sort of gasket that seals around the case of the MS5611. I think Nick Arsov did the X2 case design with the baro chamber. I'll ask him and Phil whether they can modify the one for the XRacer to add a nipple for a static port.

kd0aij, 2015-11-12:
With the X2 case modified to provide a static port nipple, I tested a static port oriented vertically in the top of a plastic dome. The AeroQuad 650 now performs similarly to the QAV250 in ALTCTL mode.
[image]

Altitude accuracy in slow flight around 2m AGL is better than +/- 2m although the autopilot thinks it's doing a lot better than that (LPOS.Z): 
[image]
http://logs.uaventure.com/view/uyLcnETynCZHNB8N7oPV8U

I wonder whether it could be tuned more aggressively, but this quad doesn't have wireless telemetry, and if I understand the new arming constraints, I can't use QGC to configure "set to RC" parameters via USB... Have there been any changes in the User Interface for set to RC to allow using it without a GCS?

SimonWilks, 2015-11-13:
​
How hard/accurate would it be to hack up a crude small wind tunnel and
place the quad I side there measuring pressure for different orientations?
We tried all these kind of things too, ok not the static port, and had
trouble really pinpointing it...too many variables when flying.

Could even make a nice student project, analysing body shapes, the pressure
zones that result in flight, effects on baro and optimal positioning/setup
of the baro, etc.

kd0aij, 2015-11-14:
I think a wind tunnel would have to be pretty large if you wanted to take propwash into account.

How accurate is the altitude reported by the Lidar Lite V1 while in forward flight around 2m AGL? Can it provide ground-truth for testing barometric altitude in flight?

SimonWilks, 2015-11-14:
Lidar: yes. We have some posted some results on this above. It gives a
pretty accurate measurement.

Wind tunnel: perhaps.

kd0aij, 2015-11-15:
Installed LidarLite V1 on the X2/AeroQuad:

[image]
log: http://logs.uaventure.com/view/7qgamGaKyjHfTeA2Mu3JfW
[image]

Looks like things are working properly. With the current tuning ALTCTL (with the pitot static probe) is losing about 1m AGL at speed according to the lidar. Roll and pitch angles are relatively small, so I'm hoping the lidar reading is accurate. I can roughly confirm the magnitude of the altitude loss from visual observation.

Note the integral of pitch is the best estimate of speed since I was flying without GPS. This platform is looking like a good testbed for static probe testing with addition of GPS for speed measurement.

kd0aij, 2015-11-16:
Test with GPS, Lidar and vertical tube for static port:
[image]

Baro error is about .8m with forward speed around 3.5m/sec
0.5m error at 2.5m/sec

kd0aij, 2015-11-17:
@AndreasAntener @LorenzMeier 

In the plot below (full log: http://logs.uaventure.com/view/EEYerTNdmZk5pTghWCXDvW) 
GPSP.Alt is set (published?) about 20m above the current altitude just after t=106.0 
while MainState is 1 (ALTCTL). 
This results in an unexpected rapid climb which I terminated with throttle at about 107.25.

Is this a bug, or a stupid developer trick? 
(beta firmware plus LidarLite on AUAV-X2 with hack to use AUX4 as GPIO output: https://github.com/kd0aij/Firmware/commit/034ed6ef78dd2938f50f40a0ab3e826925cdfe52)

[image]

LorenzMeier, 2015-11-18:
@kd0aij Any new insights regarding the static port placement yet?

kd0aij, 2015-11-18:
@LorenzMeier The sub-meter error with top-mounted vertical tube static port is the best I've seen so far for flying at speeds up to 4 m/sec. I plan to do steady-state testing at higher speeds as soon as we have good flying weather here at the field. At speeds up to 4 m/sec there is no observable difference between the vertical tube (about 3mm inside diameter) and a pitot static tube in several different locations. 

Two questions for the core team:
1) How soon will https://github.com/PX4/Firmware/issues/3188 be addressed, or would I be able to implement it?
2) Does the log in my last comment here indicate a bug in the core code, or a usage error on my part?

mhkabir, 2015-11-18:
@kd0aij I've implemented high-performance altitude hold using rangefinders in this pull : #3209 
Mind giving it a whirl? :) 
This is for the LPE filter only though.

kd0aij, 2015-11-18:
@mhkabir I haven't tried the LPE filter yet, but I'd definitely like to try your altitude hold enhancements. How would I configure my X2/Lidar-lite quad to run LPE?

mhkabir, 2015-11-18:
Check out the pull request I linked to. It's just flash my lpe_flow branch
and go. It has several changes to make sure rangefinders work as required.

Upload the lpe target with
```
make px4fmu-v2_lpe upload
```
That's it. Should work just out of the  box then.

kd0aij, 2015-11-18:
This is what I get after flashing target px4fmu-v2_lpe:
```
PID   PRI SCHD TYPE   NP STATE    NAME
    0   0 FIFO TASK      READY    Idle Task()
    1 192 FIFO KTHREAD   WAITSIG  hpwork()
    2  50 FIFO KTHREAD   WAITSIG  lpwork()
    3 100 FIFO TASK      RUNNING  init()
  100 250 FIFO TASK      WAITSEM  sensors()
    6  70 FIFO TASK      WAITSIG  nshterm(10003f50)
  103 220 FIFO TASK      WAITSEM  gps()
  168 250 FIFO TASK      WAITSEM  mc_att_control()
  105 215 FIFO TASK      WAITSIG  commander(100098c0, 100067b0)
  106  50 FIFO PTHREAD   WAITSEM  commander_low_prio(0)
  172 250 FIFO TASK      WAITSEM  mc_pos_control()
  109 240 FIFO TASK      WAITSEM  px4io()
  176 105 FIFO TASK      WAITSEM  navigator()
   82  90 FIFO TASK      WAITSEM  dataman()
  115 100 FIFO TASK      WAITSIG  mavlink_if0(100065b0, 100098b0, 10009c10, 100)
  116 175 FIFO PTHREAD   WAITSEM  mavlink_rcv_if0(2000e240)
  156 250 FIFO TASK      WAITSEM  attitude_estimator_q()
  158 250 FIFO TASK      WAITSEM  position_estimator_inav()
  127  70 FIFO TASK      WAITSIG  sdlog2(10004610, 2000cc20, 2000cc30, 2000d020)
```
Shouldn't local_position_estimator have replaced position_estimator_inav?

scott-eddy, 2015-11-18:
You need to disable INAV with INAV_Enable parameter and then the LPE will be initialized

kd0aij, 2015-11-19:
First test using Lidar to fly around: first at just under 1m AGL (120-350) then at about 2m AGL. Looks like below 1m there are some ground effects causing baro fluctuations:

[image]

But at 2m baro discrepancy is better than +/-0.5m. This is still at low speed; 3 to 4 m/sec max.

tumbili, 2015-12-02:
[image]

tumbili, 2015-12-02:
@LorenzMeier @kd0aij I just completed a altitude hold test with the XRacer board. I didn't use the gps and I increased the baro weight in INAV from 0.5 to 0.8.
What I noticed is that the baro is very sensitive to the roll rate of the vehicle. If I'd hover very still and then just apply a small impulse to my roll stick the vehicle would fall about two meters.
You can also see this in the plot above.
The horizontal velocity of the vehicle during this maneuver was very small.

LorenzMeier, 2015-12-02:
We can look at the mechanics of your setup. We certainly need to do something about your bare Pixracer board, its never going to work well in the current shape.

LorenzMeier, 2015-12-07:
We identified two things now: The missing pre-compensation and baro. Closing this now as the low-hanging fruit have been handled.

closed 2015-12-07 by LorenzMeier