PX4/PX4-Autopilot #2795: SITL startup script

tumbili, 2015-09-02:
@mcharleb @LorenzMeier At the moment we are trying to configure streams from the SITL startup script. This does not work because at that moment the mavlink instance is still waiting for the udp connection and thus is not functional yet.

LorenzMeier, 2015-09-02:
This points to the need for the waiting to be implemented differently. There is no reason for it to reject stream config commands while its not streaming anything yet.

tumbili, 2015-09-02:
@LorenzMeier I actually tried to shift the initialization of the udp further down where the instance already exists but then I don't get any parameters anymore in QGC. The streams work fine.

LorenzMeier, 2015-09-02:
Is this fixed now?

tumbili, 2015-09-02:
Yes, fixed with #2796

closed 2015-09-02 by tumbili