PX4/PX4-Autopilot #2813: RTL can lead to invalid position setpoint, which is not guarded against

LorenzMeier, 2015-09-05:
Here:
http://logs.uaventure.com/view/bcuV4d7CuG5AqhH5z3uwq

On the switch to RTL, X / Y pos and velocity become NaN:
```
2015-08-27 11:13:01.91: LPSP {X : 34.4201622009, Y : -37.3390235901, Z : -8.90506362915, Yaw : -0.747100353241, VX : 0.129600524902, VY : 0.0175743103027, VZ : 0.0788640975952, AX : 0.0280562341213, AY : 0.0294567756355, AZ : -0.425087034702}
2015-08-27 11:13:01.92: LPSP {X : 34.4201622009, Y : -37.3390235901, Z : -8.90506362915, Yaw : -0.747100353241, VX : 0.133010864258, VY : 0.0206260681152, VZ : 0.0792627334595, AX : 0.0280498098582, AY : 0.0298606157303, AZ : -0.42488014698}
2015-08-27 11:13:01.93: LPSP {X : nan, Y : nan, Z : -8.98391723633, Yaw : 2.31342053413, VX : nan, VY : nan, VZ : 0.000919342041016, AX : nan, AY : nan, AZ : -0.440413475037}
2015-08-27 11:13:01.94: LPSP {X : nan, Y : nan, Z : -8.98391723633, Yaw : 2.31342053413, VX : nan, VY : nan, VZ : 0.00150775909424, AX : nan, AY : nan, AZ : -0.440164715052}
```

tumbili, 2015-09-07:
@LorenzMeier I've been looking at the relevant code and I can conclude the following:

When the user switched to RTL the code for RTL_STATE_RETURN was executed since the copter was well over the return altitude. This means that the current position setpoint was set to the home position and altitude was set to the current altitude. Furthermore, the previous position setpoint was set to invalid.

In the mc position controller this means that only very little code in control_auto() should have been executed because of this https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L815

Anyway, I still can't see where a NaN in the position setpoint could have been created since this is guarded https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L899

Another potential source could be here https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L801 but this also looks fine to me.

LorenzMeier, 2015-11-06:
We merged the patch, so closing.

closed 2015-11-06 by LorenzMeier