PX4/PX4-Autopilot #2820: Navigator broken for POSIX

tumbili, 2015-09-07:
@LorenzMeier I think for posix SCHED_PRIORITY_DEFAULT is already defined to be the max possible priority. So the statement "SCHED_PRIORITY_DEFAULT + 5" probably brakes it.

LorenzMeier, 2015-09-07:
That would be a bug of the define. Is should be == (max - min) / 2

LorenzMeier, 2015-09-07:
Fixed in 9a30fb7.

LorenzMeier, 2015-09-07:
Please re-test.

tumbili, 2015-09-07:
@LorenzMeier Shouldn't it be min + (max - min) / 2 ?

LorenzMeier, 2015-09-07:
yep, was too hasty. Fixing.

LorenzMeier, 2015-09-07:
Done. Please re-test. Also need to look what min and max are.

tumbili, 2015-09-07:
@LorenzMeier Works

tumbili, 2015-09-07:
@LorenzMeier min is 1, max is 99

LorenzMeier, 2015-09-07:
Its +- 127 for NuttX. So there is some work needed to replace the magic numbers with defines which correctly map the ranges. I've renamed this issue and left it open.

closed 2015-11-25 by LorenzMeier