PX4/PX4-Autopilot #2822: Implement stick mapper work queue

tumbili, 2015-09-08:
The goal is to remove any non-control related attitude setpoint generation code out of the control modules. E.g. at the moment attitude setpoints in manual mode are generated in mc_pos_control and fw_att_control.
The idea is to create a stick mapper function which is implemented as a work queue in order to save stack size. This function should map manual control inputs from the user to attitude setpoints for the attitude controllers.
For the multicopter case this is straight forward because in every mode the attitude setpoint is either published by the position controller or by the stick mapper function.
For the fixed wing case this is a bit more tricky, since we have modes like ALTCTL and POSCTL in which the attitude setpoint is generated by a combination of fw position controller and stick mapper function. E.g in ALTCTL pitch_sp and thrust_sp are computed by the controller while roll_sp should be computed by the stick mapper function.

mhkabir, 2015-09-29:
@tumbili I've been working with obstacle avoidance on multicopters, and have been generating position targets for the avoidance algorithm from a joystick offboard. Eventually, I plan to simply use the stick inputs from RC converted to position targets (much like what is done by mc_pos_ctl in POSCTL currently)

Moving the stick to setpoint functions out of the controllers would thus be something I'm interested in. Would you let me know what needs to be done? :)

tumbili, 2015-10-21:
@maxb91 We can move the discussion in here.
@mhkabir We are going to come up with some sort of "proposal" on our side,  probably in the form of a PR. Then everybody can comment on it and we can further enhance it.

LorenzMeier, 2015-11-06:
@mswingtra @yyassa Where is the stick mapper standing?

mswingtra, 2015-11-06:
@maxb91 could you maybe update here on the current state?

LorenzMeier, 2015-11-18:
Any news on this?

maxb91, 2015-11-23:
We ran some first tests of the stick mapper module and they looked good. So far, the manual setpoints for mc_pos_control were implemented and the next step is to also include fw_att_control.

tumbili, 2015-11-23:
@maxb91 Nice! Could you provide the link to the branch or make a PR? The last link you sent me didn't work so make sure the branch is not private.

tumbili, 2015-11-30:
@maxb91 Ping me if you need help on this.

tumbili, 2016-02-19:
The guy who was working on this was fired maybe :-)

dagar, 2016-02-25:
I can help with this for fixed wing. Where's the current work?

LorenzMeier, 2016-02-25:
I don't think there is any real work for this. Its probably best to start fresh, in particular given your experience.

LorenzMeier, 2016-02-25:
@dagar Here is the optimal template to start with:
https://github.com/PX4/Firmware/blob/master/src/modules/land_detector/LandDetector.cpp

The update rate should depend on a param - racing people will want to run the sampling much faster at 100 or 200 Hz.

LorenzMeier, 2016-02-25:
@dagar Thanks for taking over! @julianoes Don't forget this in your review.

dagar, 2016-02-26:
I'm not sure if I properly understand the intended design with the work queue. Is the overall goal of this to centralize attitude setpoint in the position controllers, but use the separate thread to publish faster?

LorenzMeier, 2016-02-26:
No. The goal is to centralise attitude setpoint generation in its own app and the work queue avoids the overhead of creating a complete app on the stack, and instead runs in the work queue stack. Its a very modern design pattern used to be low-latency and resource efficient, also used in responsive frameworks like Node.js.

A position controller should not contain manual flight mode handling. This is the proper way to efficiently not have that in there.

dagar, 2016-04-23:
**Stick Mapper Design**

Fixed Wing

 - Manual -> publish actuator controls
 - Stabilized -> publish attitude setpoint
 - Altitude -> publish new uorb message with airspeed, altitude, and **turn rate?** setpoints consumed by fw_pos_control.
 - Position -> same as Altitude, but fw_pos_control uses l1 to hold heading 
 - Auto -> publish nothing (eventually allow manual input/adjustment during auto?)

Fixed Wing Notes 
 -  altitude mode could use l1 navigate_heading()
 - pull offsets (eg FW_PSP_OFFSET) into fw_pos_control (otherwise ATSP logging can be confusing)

Multicopter TBD
 - Manual/Stabilized -> publish attitude setpoint
 - Acro -> publish rates setpoint
 - Rattitude -> need to review
 - Altitude ->
 - Position ->
 - Auto ->

dagar, 2017-01-15:
Coming back to this with a much better understanding of the existing architecture.

Proposed Stick mapper publications:
 MANUAL: publish actuator_controls (I think MC MANUAL should actually just switch to STAB rather than fake it)
 STABILIZED: publish attitude setpoints
 ACRO: publish rate setpoint
 RATTITUDE: switch between rate and attitude setpoints based on control mode. Rattitude threshold would now cause the control flags and subsequent publications to change.
 ALTITUDE: use position setpoint for airspeed, altitude. Need someway to pass roll through, or start treating it like heading.
 POSITION: velocity and altitude for MC, FW roll pass through again

**Questions**
 - How should altitude control roll work? Roll setpoint from sticks passed through via position_setpoints or should the position controller continue to take this directly from manual?
 - Need to review use of manual flaps, gear, aux{1-5} and get them out of the controllers

LorenzMeier, 2017-10-04:
Closing as stale.

closed 2017-10-04 by LorenzMeier