PX4/PX4-Autopilot #2823: Velocity control with OFFBOARD mode failed

weiweikong, 2015-09-08:
Hi, 
    I am working with the velocity control with OFFBOARD mode. My setup are
- jMAVSim
- QGroundcontrol
- Pixhawk firmware-stable

After checking some issues, such as
- https://groups.google.com/forum/#!searchin/px4users/velocity$20control/px4users/LFzy9Vaoimg/in9bdrDTSisJ
- https://groups.google.com/forum/#!searchin/px4users/velocity$20control|sort:date/px4users/czVQi7EmgGY/Efa7DlucAAAJ
- https://github.com/PX4/Firmware/issues/1238

I am stll confusing about

## The velocity controller still not allow in Pixhawk firmware-stable?
I published the `/mavros/setpoint_velocity/cmd_vel` for testing, with stable speed in each direction, such as (vx = 1, vy = 0, vz = 0), (vx = 0, vy = 1, vz = 0). The quadrotor flew to the wrong direction with high speed then slow down. When I tried the (vx=0, vy=0, vz=0), the quadrotor flew randomly near the setpoint, but not keep stable as `AUTO|LOITER` mode. But I check the `ms_pos_control_main.cpp` the function `control_offboard` does some work relate to control.
- Is the velocity controller not working ? or the method of publishing OFFBOARD method is run.
- Which way is the standard one to allow OFFBOARD velocity control ?
- Also, in PX4 ROS SITL, https://github.com/mavlink/mavros/tree/master/test_mavros, I got successful result that the system could work under OFFBOARD velocity control with gazebo. But when I tried to run the same publish node to Pixhawk, it still failed. Is the velocity controller in PX4 ROS SITL not the same as the one used in Pixhawk firmware-stable ?

## Tthe data transfer flow
To find the result, it seems that the velocity setpoints transfer as,
Step 1: from `/mavros/setpoint_velocity/cmd_vel` to `MavlinkReceiver::handle_message_set_position_target_local_ned(mavlink_message_t *msg)` in `mavlink_receiver.cpp` 
Step 2: then applied by `MulticopterPositionControl::control_offboard(float dt)` in `ms_pos_control_main.cpp`
- Is this process right ?

## Compiling Issue
 I am just wondering the issue that there is no `uORB/topics/position_setpoint_triplet.h`, also other related head files, in Firmware-stable(https://github.com/PX4/Firmware/tree/master/src/modules/uORB), but I could find those in Firmware-RTL branch (https://github.com/PX4/Firmware/tree/RTL/src/modules/uORB). 
- Why the `catkin_make` also complie the packages correctly, when `ms_pos_control_main.cpp` include the head file, but there is no such head file in related directory?

Thanks.

vooon, 2015-09-08:
uORB topic types are generated from [ROS IDL files](https://github.com/PX4/Firmware/tree/master/msg).

weiweikong, 2015-09-08:
Thanks @vooon ! Get the point !

v01d, 2015-10-17:
@weiweikong I'm not quite sure about your problem, but current master should work better for velocity setpoints. If you consider it fixed, it might be best to close this.

closed 2016-11-19 by LorenzMeier