PX4/PX4-Autopilot #2824: SITL floating point exception

tumbili, 2015-09-08:
roman@roman-System-Product-Name:~/Firmware$ make sitl_plane
INFO  Shell id is 47044602642176
App name: mainapp
pxh> arg 0 simulator
arg 1 start
arg 2 -s
INFO  Simulator started
WARN  0          failed to open /dev/ttyACM0 (file /home/roman/Firmware/src/modules/simulator/simulator_mavlink.cpp line 404)
WARN  12         failed to send streaming command to /dev/ttyACM0 (file /home/roman/Firmware/src/modules/simulator/simulator_mavlink.cpp line 412)
WARN  18         Waiting for initial data on UDP.. Please connect the simulator first. (file /home/roman/Firmware/src/modules/simulator/simulator_mavlink.cpp line 428)
WARN  28856117   Found initial message, pret = 1 (file /home/roman/Firmware/src/modules/simulator/simulator_mavlink.cpp line 432)
WARN  6          Sending initial controls message to jMAVSim. (file /home/roman/Firmware/src/modules/simulator/simulator_mavlink.cpp line 439)
pxh> WARN  135354     open 'rootfs/eeprom/parameters' (file /home/roman/Firmware/src/systemcmds/param/param.c line 247)
pxh>   MAV_TYPE: unchanged
pxh>   SYS_AUTOSTART: curr: 0 -> new: 3033
pxh>   SYS_RESTART_TYPE: unchanged
pxh> dataman: Unknown restart, data manager file 'rootfs/fs/microsd/dataman' size is 101504 bytes
pxh>   CAL_GYRO0_ID: curr: 0 -> new: 2293760
pxh>   CAL_ACC0_ID: curr: 0 -> new: 1376256
pxh>   CAL_ACC1_ID: curr: 0 -> new: 1310720
pxh>   CAL_MAG0_ID: curr: 0 -> new: 196608
pxh>   CAL_GYRO0_XOFF: curr: 0.0000 -> new: 0.0100
pxh>   CAL_ACC0_XOFF: curr: 0.0000 -> new: 0.0100
pxh>   CAL_ACC0_YOFF: curr: 0.0000 -> new: -0.0100
pxh>   CAL_ACC0_ZOFF: curr: 0.0000 -> new: 0.0100
pxh>   CAL_ACC0_XSCALE: curr: 1.0000 -> new: 1.0100
pxh>   CAL_ACC0_YSCALE: curr: 1.0000 -> new: 1.0100
pxh>   CAL_ACC0_ZSCALE: curr: 1.0000 -> new: 1.0100
pxh>   CAL_ACC1_XOFF: curr: 0.0000 -> new: 0.0100
pxh>   CAL_MAG0_XOFF: curr: 0.0000 -> new: 0.0100
pxh>   MPC_XY_P: curr: 1.0000 -> new: 0.4000
pxh>   MPC_XY_VEL_P: curr: 0.1000 -> new: 0.2000
pxh>   MPC_XY_VEL_D: curr: 0.0100 -> new: 0.0050
pxh>   COM_RC_IN_MODE: curr: 0 -> new: 2
pxh> WARN  1576520    could not open /dev/rgbled0 (file /home/roman/Firmware/src/drivers/device/i2c_posix.cpp line 125)
WARN  1576535    no RGB led on bus #3 (file /home/roman/Firmware/src/drivers/rgbled/rgbled.cpp line 641)
pxh> INFO  TONE_SET_ALARM 0
pxh> Invalid command: gyrosim
type 'help' for a list of commands
pxh> pxh> pxh> pxh> gps lock
gps unlock
pxh> WARN  1987482    usage: meas_airspeed_sim command [options] (file /home/roman/Firmware/src/platforms/posix/drivers/airspeedsim/meas_airspeed_sim.cpp line 570)
WARN  1987492    options: (file /home/roman/Firmware/src/platforms/posix/drivers/airspeedsim/meas_airspeed_sim.cpp line 571)
WARN  1987494    	-b --bus i2cbus (1) (file /home/roman/Firmware/src/platforms/posix/drivers/airspeedsim/meas_airspeed_sim.cpp line 572)
WARN  1987496    command: (file /home/roman/Firmware/src/platforms/posix/drivers/airspeedsim/meas_airspeed_sim.cpp line 573)
WARN  1987498    	start|stop|reset|test|info (file /home/roman/Firmware/src/platforms/posix/drivers/airspeedsim/meas_airspeed_sim.cpp line 574)
pxh> pwm_out_sim MODE_4PWM
pwm_out_sim starting
pxh> pxh> WARN  3184045    FATAL: no gyro found: /dev/gyro0 (file /home/roman/Firmware/src/modules/sensors/sensors.cpp line 941)
WARN  3184058    Sensor initialization failed (file /home/roman/Firmware/src/modules/sensors/sensors.cpp line 2123)
WARN  3184129    start failed (file /home/roman/Firmware/src/modules/sensors/sensors.cpp line 2342)
pxh> arg 0 commander
arg 1 start
WARN  3249339    No RGB LED found at /dev/rgbled0 (file /home/roman/Firmware/src/modules/commander/commander_helper.cpp line 301)
INFO  IOCTL DEVIOCGDEVICEID 196608
INFO  IOCTL DEVIOCGDEVICEID 1310720
telem> PREFLIGHT FAIL: ACCEL #0 UNCALIBRATED
telem> PREFLIGHT FAIL: NO GYRO SENSOR #0
telem> PREFLIGHT FAIL: AIRSPEED SENSOR MISSING
INFO  TONE_SET_ALARM 9
INFO  ToneAlarm::start_tune
INFO  ToneAlarm::start_note 8522
telem> Not ready to fly: Sensors need inspection
pxh> WARN  3450945    INVAL: ARMING_STATE_STANDBY_ERROR - ARMING_STATE_STANDBY (file /home/roman/Firmware/src/modules/commander/state_machine_helper.cpp line 271)
pxh> pxh> INFO  ToneAlarm::start_note 8522
INFO  .
INFO   
WARN  3623228    timed out (file /home/roman/Firmware/src/modules/navigator/navigator_main.cpp line 340)
pxh> .
pxh> INFO  ToneAlarm::start_note 8522
.
pxh> pwm_out_sim lock
pwm_out_sim unlock
pwm_out_sim lock
pwm_out_sim unlock
pwm_out_sim lock
pwm_out_sim unlock
pxh> arg 0 mavlink
arg 1 start
arg 2 -u
arg 3 14556
arg 4 -r
arg 5 60000
WARN  3987465    mode: 0, data rate: 60000 B/s on udp port 14556 (file /home/roman/Firmware/src/modules/mavlink/mavlink_main.cpp line 1539)
WARN  3987576    offboard mission init: ERROR (file /home/roman/Firmware/src/modules/mavlink/mavlink_mission.cpp line 134)
INFO  Setting up UDP w/port 14556

WARN  4023305    timed out (file /home/roman/Firmware/src/modules/navigator/navigator_main.cpp line 340)
pxh> WARN  4053097    stream POSITION_TARGET_LOCAL_NED on UDP port 14556 enabled with rate 50.0 Hz (file /home/roman/Firmware/src/modules/mavlink/mavlink_main.cpp line 1807)
INFO  ToneAlarm::start_note 10134
pxh> WARN  4123471    stream LOCAL_POSITION_NED on UDP port 14556 enabled with rate 50.0 Hz (file /home/roman/Firmware/src/modules/mavlink/mavlink_main.cpp line 1807)
pxh> WARN  4193422    stream ATTITUDE on UDP port 14556 enabled with rate 50.0 Hz (file /home/roman/Firmware/src/modules/mavlink/mavlink_main.cpp line 1807)
WARN  4223399    timed out (file /home/roman/Firmware/src/modules/navigator/navigator_main.cpp line 340)
pxh> WARN  4263704    stream ATTITUDE_TARGET on UDP port 14556 enabled with rate 50.0 Hz (file /home/roman/Firmware/src/modules/mavlink/mavlink_main.cpp line 1807)
pxh> pxh> arg 0 sdlog2
arg 1 start
arg 2 -r
arg 3 100
arg 4 -e
arg 5 -t
arg 6 -a
WARN  4398900    ERR: open in (file /home/roman/Firmware/src/modules/sdlog2/sdlog2.c line 2030)
WARN  4398906    unable to copy conversion scripts (file /home/roman/Firmware/src/modules/sdlog2/sdlog2.c line 1085)
WARN  4398909    log buffer size: 8192 bytes (file /home/roman/Firmware/src/modules/sdlog2/sdlog2.c line 1091)
telem> [sdlog2] log dir: rootfs/fs/microsd/log/2015-09-
telem> [sdlog2] starting: 14_07_26.px4log
WARN  4423387    timed out (file /home/roman/Firmware/src/modules/navigator/navigator_main.cpp line 340)
Tools/sitl_run.sh: line 5:  7243 Floating point exception(core dumped) ./mainapp ../../$1
make: *** [sitl_plane] Error 136
roman@roman-System-Product-Name:~/Firmware$

LorenzMeier, 2015-09-08:
@mcharleb My first preference would be a handler printing where the NaN happened, my second preference would be to set ```-fno-trapping-math, -fno-signaling-nans``` to disable this.

tumbili, 2015-09-08:
After recompiling it usually works.

mcharleb, 2015-09-08:
@tumbili That just means you were (un)lucky if it didn't happen again.

@LorenzMeier The following link shows how to code for handling C++ floating point exceptions:

http://en.cppreference.com/w/cpp/numeric/fenv/feclearexcept

I tried in on Linux and it works as expected (g++ 4.9 using -std=c++11). It also works with clang-3.6 but clang warns about the unsupported pragma.

LorenzMeier, 2015-09-08:
Yeah, but I now remembered that we're using NaNs also to represent invalid fields and deal with it by proper checks. So we will have to go down the route of disabling exceptions.

devbharat, 2015-09-13:
Can you link to the commit that resolved this issue so I can pick it, I am on a slightly older version and this issue occurs persistently.

devbharat, 2015-09-13:
https://github.com/PX4/Firmware/pull/2798 Resolved it for me, thanks!

LorenzMeier, 2015-09-21:
Duplicate of #2802

closed 2015-09-21 by LorenzMeier