PX4/PX4-Autopilot #2830: offboard velocity setpoint not responding in XY

v01d, 2015-09-09:
I have discussed the problem, which has been acknowledged, in this thread: https://groups.google.com/d/msg/px4users/czVQi7EmgGY/we8feV9KAAAJ

The problem is present in these lines: https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1088-L1092

The if condition depends con "position control" not being enabled but the result is the reset of the velocity setpoint (computed in the lines just above). With offboard velocity setpoint, the "position control" is disabled but "velocity control" is enabled. I understand this is correct as per this description:

https://github.com/PX4/Firmware/blob/c62ae87c69963a4880c990c02000fd09e9b228f4/msg/vehicle_control_mode.msg

Thus, I suggest changing the condition to test if "velocity control" is disabled, not "position control".

However, on the mailing list another fix was suggested (for which I'm not really sure it is correct), where under velocity setpoint mode, the "position control" is enabled. However, this feels contrary to the point of these two booleans.

Another suggestion was to go forward with actual velocity control (not the "rate of position setpoint movement" control which is implemented at present, as proposed in this PR:  https://github.com/PX4/Firmware/pull/2305

While I like the idea of the PR I feel a more "urgent" fix is needed to the stable branch (currently affected) for which I have succesfully tested the proposed fix. The PR still does not address offboard mode (only manual mode) so I think this has still some time to be finished.

LorenzMeier, 2015-09-20:
I'm fine with an incremental approach. Would you like to set up a call some time this week to coordinate?

v01d, 2015-09-20:
Sorry, what do you mean by "incremental approach"?

LorenzMeier, 2015-09-20:
We can try to apply a hotfix while we work together on a more fundamental "proper" solution.

v01d, 2015-09-20:
Ah, ok. I can provide the pull request if you want, it is more or less trivial.

Regarding the offboard control itself, we can discuss further. However, my knowledge of the general codebase is mostly vague. I'm still trying to learn by looking at the code.

weiweikong, 2015-09-25:
Hi @v01d ,
   I am also interesting in the control based on the velocity setpoints.

### About the code fix

 When you modifeid the code
from
```
if (!_control_mode.flag_control_position_enabled) 
```
to
```
if (!_control_mode.flag_velocity_position_enabled) 
```
would it be safe when drones do not received valid position setpoint ?
According to this fix, the drones do not set `_vel_sp(0) = 0` and `_vel_sp(1) = 0`, which would be dangeroud ?

### About the velocity scale

When we set the velocity setpoint, how could we decide the scale and unit of these setpoints. For example, when we set `linear.x=1.0` with `/mavros/setpoint_velocity/cmd_vel`, does this mean the expected velocity is 1.0 m/s in the body frame ?

v01d, 2015-09-25:
@weiweikong please have a look at the description in my first post where I explained the problem. This only affects offboard mode.

With current implementation, you don't control velocity directly but the speed at which a position setpoint is moved ahead ("carrot follower"). If drone behaves well, the speed should be more or less constant, but it will depend on PID constants. If the drone lags behind, it would speed up. In any case, there are maximum speed limits. There's a pull request (linked in my first post) where this behavior is changed.

weiweikong, 2015-09-26:
@v01d  I checked the `mc_pos_control_main.cpp`, in the [line 1041](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1041),
```
if (!_control_mode.flag_control_manual_enabled && _pos_sp_triplet.current.valid && _pos_sp_triplet.current.type == position_setpoint_s::SETPOINT_TYPE_IDLE) {
```
and
the previous code
```
if (!_control_mode.flag_control_position_enabled) {
					_reset_pos_sp = true;
					_vel_sp(0) = 0.0f;
					_vel_sp(1) = 0.0f;
				}
```
runs in the `else` section. In my opinion, except `idle state`, all the rest control modes will go through `else` section, so the modified section might be called. 

Thank you for the clear description about the `velocity control` definition which I was not catch the key idea of this kind of path following method previously.

weiweikong, 2015-09-28:
Some update about the OFFBOARD velocity control and position control. 

## 1. Basic Setup
- Pixhawk with modified firmware.
The code was only changed from `if(!_control_mode.flag_control_position_enabled) ` to `if(!_control_mode.flag_velocity_position_enabled) `
- Jetson TK1 as the companion computer
- All the OFFBOARD posotion/velocity setpoints send from TK1 to Pixhawk through Tele2 port.

## 2. OFFBOARD position control
### Trajectory
[image]
### Local Position X
[image]
### Local Position Y
[image]
### Local Position Z
[image]

The position setpoints were sending until the Quadrotor reach the demanding point, so the green line in X and Y axis were zigzag. Also, the behavior of the Quadrotor was very standard, step by step, which **hovered at each setting points** in the circle for a while.

## 3. OFFBOARD velocity control
### Trajectory
[image]
### Local Position X
[image]
### Local Position Y
[image]
### Local Position Z
[image]

The velocity setpoints were updating while the Quadrotor was flying, so the green line in X and Y axis were **much more smoothly** than position-based control. Without checking whether the Quadrotor reached the point on the circle, the red line was rise and fall near the desired trajectory. However, there was **no hovering maneuver**. 

Above are some tests I did, it might be right and stable after modifying the firmware. @v01d  @LorenzMeier, I am still new to Pixhawk, so some suggestions and further test roadmap are welcome.

v01d, 2015-09-28:
Hi, from what I understand you made the suggested change and performed some
test flights. Did you also test without the change? The problem I had
simply did not allow velocity control, it was not a matter of smoothness
(that depends on PID parameters).

weiweikong, 2015-09-28:
@v01d  Yes, I changed the firmware as you suggestions. 

Without changing, I also tested the velocity offboard control with real flight and jMAVSim based HIL and both of these two setup could NOT get the correct behivour no matter what kind of velociy setpoints sending to Pixhawk. 

I did not catch the point of your last point. What do you mean by 'not allow velociy control' ?

v01d, 2015-09-28:
I mean it did not even respond to X,Y velocity setpoints. The response was
about 100 times smaller and I believe this was because it would immediately
zero the velocity setpoint after sending it.

weiweikong, 2015-09-28:
@v01d I tried to increase the velocity setpoints from 0.1 to 200 in real flight and HIL. When the velocity setpoint in X- or Y-axis bigger than 100, the quadrotor can move to the desired direction. However, it stopped even though the velocity setpoints stream was still sending. Then, I tried to change to the other axis, the quadrotor just stopped there with random, in my mind, moving near the last point. 

- Previously, there were some discussion in google group, https://groups.google.com/forum/#!msg/px4users/czVQi7EmgGY/we8feV9KAAAJ, but I did not get the final solution from that.

- The only setup, which could implement the velocity offboard control correctly, is based on Gazebo, ROS and MAVROS, https://github.com/mavlink/mavros/tree/master/test_mavros. Based on this setup, I could run the velocity offboard control based on the stable Pixhawk firmware. But, I check the details of the launch file, which simply start the `ms_pos_control` module. In my opinion, this module is the same one that running in real flight. So I am confused with that and I opened a issue there, https://github.com/mavlink/mavros/issues/384, and have not received clearly replay and suggestions yet. 

- Also, I tried to add some `mavlink_log_info` to debug the `ms_pos_control`, checking and outputting the variable details while the position control loop was running, but I could not receive any outputs  through mavlink and mavros both from terminal and QGC. So I did not know which is the better way to debug the detail in this module with ROS STIL or jMAVSim based HIL setup.

v01d, 2015-09-28:
>    opened a issue there, mavlink/mavros#384

Ok, I understand than that you confirm my previous problem.

Regarding Gazebo simulation, I don't think it is the same code, it should
actually be mc_pos_control_multiplatform. But maybe someone can confirm
that.

In any case, this is just a quick fix. There's another effort going on to
switch to proper velocity control.

v01d, 2015-09-28:
Just to complete the analysis: I was looking at the code in order to implement actual velocity control and found this line:

https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1151

To me, this explains the previous behavior. In offboard mode vel_sp was always being set to zero. However, if you set velocity v=100, it would respond with v=1. The cause is most likely due to this line, where D error es computed not from vel_sp (which was 0) but from sp_move_rate (the input from the user). Since D is by default 0.01 this explains the behavior.

The question is why is D computed from sp_move_rate in this case and not from vel_sp. To me this is not correct (and it would actually be a problem for pure velocity control).

v01d, 2015-10-17:
Closing this, since #3004 supercedes this.

closed 2015-10-17 by v01d