PX4/PX4-Autopilot #2835: Multicopter attitude hold performance

tumbili, 2015-09-11:


v01d, 2015-09-14:
Probable typo on issue title, it says "attitude" instead of "altitude"

SimonWilks, 2015-09-29:
@tumbili has there been any progress on this front? 

#2790 is relevant to this discussion as well.

tumbili, 2015-09-29:
@SimonWilks The problem is clearly the barometer. And there is no way of shielding it perfectly such that it's not affected by wind any more. So we need to look at how to teach the estimator not to trust the baro as much in such situations.

v01d, 2015-09-29:
If you're interested, I've been working on my fork on a variation of INAV
using sonar for altitude hold. I had to also remove the ground detection
since it wasnt reliable. Current INAV trusts too much on baro (which,
besides noise, has considerable drift).

I've worked off stable since I needed that for reliability.

https://github.com/v01d/Firmware/pull/2

In any case, I agree the way forward is with the EKF being tested. I guess
INAV will be phased out right? Otherwise, if there's any interest in my
code, let me know.

closed 2015-11-10 by LorenzMeier