PX4/PX4-Autopilot #2844: FW auto landin with range finder

tumbili, 2015-09-12:
Tested auto landing with lidar lite range finder.
In general auto landing worked but the terrain altitude estimate from the ekf stays constant at the inititial value. Looks like the range state of the ekf is always zero.
Is there some parameter which I have overseen?

Log:
http://logs.uaventure.com/view/LRaKGWPLiPz6e34wwsi3jn

LorenzMeier, 2015-09-12:
Cool, thanks for testing! I guess the next step is then to do more bench tests with the EKF and work out what the bottom distance estimate is doing.

tumbili, 2015-09-15:
[image]

tumbili, 2015-09-15:
The upper plot shows a problem which can occur during fixed wing auto landing.
At a certain distance before the landing waypoint FW_LND_HHDIST the plane should stop navigating waypoints but rather lock to a constant heading. However, I had several flights where this distance (15m) was never reached and the plane started to roll abruptly at some point (see picture below). I'm very sure that the plane never entered heading hold mode because the mavlink message here https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L1223 never appeared in QGC.

tumbili, 2015-09-15:
[image]

tumbili, 2015-09-15:
[image]

tumbili, 2015-09-15:
You can see that the landing occurred between waypoint 4 and 5. You can also see that the plane passed the landing wapoint on the side and then tried to navigate to it after, at least that's the only thing I can imagine.

tumbili, 2015-09-15:
This is the corresponding mavlink text:
'''
TAKEOFF DETECTED
Flight mode: ALTCTL
Flight mode: AUTO_MISSION
offboard mission now running
#audio: Landing, on slope
#audio: Landing, limiting throttle
#audio: Landing, flaring
Flight mode: ALTCTL
Flight mode: STAB
LANDING DETECTED
DISARMED by safety switch
[cmd] arming state: STANDBY
[sdlog2] logging stopped
data link #0 lost
'''

tumbili, 2015-09-15:
[image]

tumbili, 2015-09-15:
In the upper plot you can see that the plane was commanded to a high roll angle during the final landing approach. This bug is fixed in #2868

SimonWilks, 2015-09-15:
Just out of curiosity why did it want to roll? Did it overshoot and want to go around? Once it is at the height it was it's too late to do anything about it anyway so limiting it makes sense.

tumbili, 2015-09-15:
@SimonWilks When the plane reaches FW_LND_HHDIST it computes to target heading to be the bearing between the previous and the landing waypoint. If this heading does not match the yaw angle of the plane at that instant it will try to correct by rolling. It's clearly visible in the upper plot.

tumbili, 2015-09-15:
BTW, I misused the ATSP.q variables to capture some interesting variables which were not logged.
So don't get confused why e.g.  the distance to the landing waypoint is ATSP.qw :-)

thomasgubler, 2015-09-15:
when the terrain estimate is working again enable the  FW_LND_USETER param which is set to 0 in your log

tumbili, 2015-09-15:
@thomasgubler I am actually using a small terrain estimator module. I temporarily adapted the fw_pos controller to use it's terrain estimate, so even though the parameter is zero, it's using the estimate. The estimate is logged in GPOS.TALT (it's the distance from the plane to the ground.) When I'm happy with the landings I will try to integrate the estimation part into the ekf. To be honest I have no idea what they are doing there, so I decided to write one from scratch.

tumbili, 2015-09-15:
This is the branch I'm using https://github.com/tumbili/Firmware/tree/fw_lidar_landing

tumbili, 2015-09-16:
This is a new log from a flight I did today:
http://logs.uaventure.com/view/268qnVvnnJq6wYQHV7hg5a
The problem with the unexpected roll is gone after the fix.
Landing is nice, the tuning of the pitch controller could be better. Terrain estimation with the lidar works great.

LorenzMeier, 2015-09-16:
Nice! I think we generally need better pitch defaults. The default tuning is always terrible. Could you spend some time tuning that on your next test run?

tumbili, 2015-09-20:
@LorenzMeier  @thomasgubler This is now tested very extensively.  They key to a good landing behaviour is a well tuned attitude and TECS controller. I think I could spend some time tuning the TECS controller (also based on the APM guide) because if altitude is not tracked well the plane will over or undershoot the landing waypoint.
One other thing that we also might look at is the handling of aborted landings. In case of an aborted landing the navigator should reset to the waypoint before the landing waypoint. Right now I think the plane tries to target the landing waypoint directly which is not a good idea.

AndreasAntener, 2015-10-09:
This can be closed together with #2964 when done.

closed 2015-10-29 by tumbili