PX4/PX4-Autopilot #2852: UAVCAN GNSS broken in master

mhkabir, 2015-09-13:
Cannot get any GPS lock in master. Native serial GPS driver was also disabled. 

Stable works.

mhkabir, 2015-09-13:
@pavel-kirienko

davids5, 2015-09-13:
@mhkabir - There was an update to UAVCAN - There is new FW that supports it. Please check the git version of uavcan stable most likely predates that change. @pavel-kirienko may be AFK but check with him.

SimonWilks, 2015-09-13:
Hi Kabir

We are using the UAVCAN GNSS on last weeks master. As David mentioned we
did have to update the FW, though I thought it wasn't working on stable or
master at all till I did that. You say you are only failing to get a GPS
lock, does the GNSS status look good when you run 'uavcan status'? I was
getting a receive N/A (or something similar) until I updated the firmware.

Simon

mhkabir, 2015-09-13:
David, Simon, yes. I'm aware of the GNSS firmware update/uavcan updates. I've been working on this since before any of that and even managed to semi-brick 2 GNSS boards before we recovered them.
I'm fully synced with Zubax master firmware. And yes, the GNSS is communicating with UAVCAN. ```uavcan status``` shows proper output. Dynamic node ID is allocated, and the mag works fine.
Me and Pavel already had a go at troubleshooting this to no avail. 
Also @Aperture-Laboratories can reproduce.

SimonWilks, 2015-09-13:
Did it work on master for you at one point and now doesn't? I can possibly
help seeing we have gnss working on last week's master but I'm now not sure
what is that you want or if you are asking for help or not. So if i can
help test something let me know.

Aperture-Laboratories, 2015-09-13:
It worked on master about a week and half ago and doesn't work after we applied https://github.com/PX4/Firmware/pull/2818 to master. Could you try adding that PR to your current build of master to see if it breaks GNSS capabilities?

SimonWilks, 2015-09-14:
Hi Bryan

Sure, I'll be able to try that out in a couple of hours. If I can reproduce
it I'll follow up with Lorenz and test a fix.

Simon

mhkabir, 2015-09-14:
Actually I can't seem to get a lock even with verbatim master. In any case, Simon, please let us know if you can reproduce with/without the PR with master as base.

pavel-kirienko, 2015-09-14:
I'll investigate the issue next Monday, unless @mhkabir or @Aperture-Laboratories solve it sooner. Zubax GNSS has been tested with PX4 version 1.0 and it worked fine, probably there was a regression in master.

sjwilks, 2015-09-14:
I merged #2818 with the version of master we were running and flashed that
and GPS works and I get a fix. The latest master (flashed via QGC) works
for me as well. Have you guys tried flashing master via QGC to rule out any
submodule issues?

mhkabir, 2015-09-14:
Argh. I just realised. I had switched my MAV's wireless link frequency as a part of some tests some days back, and it was running right on the GPS L1 frequency.
Now switched it back to 2.3Ghz (which is usually runs at) , and all is fine. I can get a clean lock on master + #2818. Now it locks to 14-15 sats near a window within a minute.
Apologies for all the confusion. 

Now Bryan. Let's see what's wrong with your setup. Can you try going fully outdoors and giving it a good 5-6 mins to lock?

closed 2015-09-19 by mhkabir

pavel-kirienko, 2015-09-19:
Has the issue been resolved for Bryan too?