PX4/PX4-Autopilot #2864: [nuttx] ARMv7-M Interrupt Stack Bugfix

nghiaho12, 2015-09-14:
From Nuttx mailing list:

This is an important bug fix. In all "lazy" versions of the ARMv7-M interrupt handler, there is an incorrect code sequence. It has been there forever and has not, until now caused any known issue. But looking at the code, it is clearly wrong and I can't imagine how it actually worked in the past.

Here is the change:

- str r1, [sp, #-4]! /* Save the MSP on the interrupt stack */
- bl up_doirq /* R0=IRQ, R1=register save (msp) */
- ldr r1, [sp, #+4]! /* Recover R1=main stack pointer */
+ push {r1} /* Save the MSP on the interrupt stack */
+ bl up_doirq /* R0=IRQ, R1=register save area on stack */
+ pop {r1} /* Recover R1=main stack pointer */

In the previous version, the "ldr" instruction was loading from the wrong address. See https://bitbucket.org/nuttx/arch/commits/e97c72f3ce29b8a776b7e47c21281b3a7d5a6fbc for the complete change.

I recommend that anyone using ARMv7M with CONFIG_ARCH_INTERRUPTSTACK > 0 incorporate this change immediately.

Thanks to Stefan Kolb for pointing this out to me.

There is another issue in this same logic: There should be additional logic to assue that the stack is aligned to double word boundaries before calling up_doirq().

Greg

LorenzMeier, 2015-09-14:
@davids5 Could you please have a look at this? It seems like its rather high prio, at least until we've assessed it.

LorenzMeier, 2015-09-20:
Thanks for reporting. Turns out we are not affected.

closed 2015-09-20 by LorenzMeier